dernasherbrezon/r2cloud #30: Configure timezone

dernasherbrezon, 2017-10-11:
Separate button to detect using javascript

ChadOhman, 2020-01-17:
What is the timezone currently set to? I have the raspberry pi set to UTC-7 but my passes predicted via Orbitron do not match with what r2cloud is reporting.
[image]
[image]

Thanks

dernasherbrezon, 2020-01-17:
r2cloud has min and guaranteed elevation limits while calculating passes. Default values are:

- min = 8
- guaranteed = 20

That means any passes without elevation < 20 are not considered. Also observations starts/ends at 8 degrees of elevation.

Regarding your example, what is the timezone for "Time - LOC" ? What is "Elv" (is it max, or initial or)?

ChadOhman, 2020-01-17:
LOC is UTC-7 and Elv is max.

Thanks for your insight.

dernasherbrezon, 2020-03-07:
Yes, Looks like orbitron "NOAA 19 2020-01-17 05:32:39" matches r2cloud on 12:27. 5 minute difference might arise from:

1. stale TLE in r2cloud
2. min elevation in r2cloud > min elevation in orbitron.

closed 2020-03-07 by dernasherbrezon