dernasherbrezon/r2cloud #34: Generate waterfall images for weather satellites

dernasherbrezon, 2017-12-05:
Will help with debugging

closed 2017-12-24 by dernasherbrezon

zeroby0, 2020-07-14:
Hi! Can you zoom in the spectrogram to the center frequency? Currently it's multiple MHz wide, so the actual signal is just a thin line.

dernasherbrezon, 2020-07-14:
Do you have an example? Currently it is 240khz or 288khz wide. And you can't "zoom" because FFT was created once and all buckets are pixels. So if you want to zoom into the part of the spectrogram, you have to define smaller FFT bucket and run FFT on the original data.

zeroby0, 2020-07-14:
Here is a spectrogram for a NOAA 18 pass.

I thought the thin line in the center is the NOAA transmission, but if the current width is 240Khz, then the 40KHz NOAA signal is too thin to be the thin center line. If the width is 240Khz, then there is no need to zoom in more. 

But isn't this spectrogram too symmetric? That's a bit suspicious, do you know what's going on here?

[image]

dernasherbrezon, 2020-07-14:
Hey, couple of things:

1. The line right in the middle is [dc offset](https://en.wikipedia.org/wiki/DC_bias).
2. The actual signal is within thick lines on both sides. It doesn't have clear APT spikes like here, because of small spectrogram resolution.

[image]

3. They are symmetric because NOAA signals are fm demodulated and stored in .wav file. They don't have imaginary part, thus symmetric after FFT.

zeroby0, 2020-07-30:
If they are symmetric, can we drop one half so we can have a better use of space? Maybe also a smaller spectral width for NOAA satellites so we can see it better?