dernasherbrezon/r2cloud #37: Ability to change flags used to start dump1090

unixpunk, 2018-03-11:
Hi, thanks for the work on this project!  I'm interested if I can change the command line flags for dump1090?  I tried editing the defaults file for dump1090 and in .r2cloud with no luck yet.  Thanks in advance!

dernasherbrezon, 2018-03-11:
Hi!

Currently all arguments to dump1090 are handcrafted. Changing any of them might break the integration. Here is the list of supported:

> "--raw", "--net", "--quiet", "--ppm"

What command line flags do you want to add?

unixpunk, 2018-03-11:
I would be looking to add some/all of these:  --phase-enhance --modeac --fix --aggressive --mlat --net-ro-size 500 --net-ro-rate 5

I use net-beast on port 30003, but I think that's enabled by default with --net on this fork of dump1090?   Thanks!

dernasherbrezon, 2018-03-12:
I've added support for additional args:

> rx.adsb.additional.args=--modeac --fix --mlat --net-ro-size 5

However **--phase-enhance**, **--net-ro-rate 5** and **--aggressive** aren't supported by current dump1090 version.

closed 2018-03-12 by dernasherbrezon