dernasherbrezon/r2cloud #55: Any way to view the recorded sound files from the noaa satellites

Blackbeard25, 2019-06-18:
i was curious if there is a way to access the sound files for the noaa weather satellites as i was wanting to put them through wx to img on my pc to get different outputs and a image of both channels. any help?

dernasherbrezon, 2019-06-18:
Hi,

You could try downloading audio from satnogs:
https://network.satnogs.org/observations/750939/

Or websdr:
http://websdr.org

closed 2019-06-18 by dernasherbrezon

ChadOhman, 2020-06-09:
+1 for this - I'd love to be able to pull down the captured wave file for my own post-processing.

dernasherbrezon, 2020-06-09:
Is it a one time action or you want to do it regularly? You can download them manually from RPi. They are under ~/r2cloud/data/satellites/noradid/data/observationId/output.wav

ChadOhman, 2020-06-09:
Thanks again @dernasherbrezon. I'm seeing the satellites indexed here:
```
pi@satpi:~/r2cloud/data/satellites $ ls
25338  39430  42017  43137  43189  43786  43804  43881  44103  44832  45119
28654  39444  42784  43186  43466  43792  43814  43908  44365  44878  45263
32789  40069  42829  43187  43743  43798  43855  44045  44406  44885  45598
33591  41460  43017  43188  43770  43803  43880  44083  44830  45115
```
With each observation catalogued under that with the output.wav:
```
pi@satpi:~/r2cloud/data/satellites $ ls 25338/data/1591712357968/
meta.json  output.wav  spectogram.png
```
Is it a possible feature request to have that output.wav linked in the web interface?

Cheers

dernasherbrezon, 2020-06-09:
Yes, sure.

dernasherbrezon, 2020-06-13:
Fixed. Should be available after auto-update
[image]
.

ChadOhman, 2020-06-13:
Wow! I just checked and I'm already updated. Incredible.
Thank you for this! :)

[image]