dernasherbrezon/r2cloud #57: Having trouble getting started with r2cloud.txt

jeffreypkelly, 2019-10-25:
HI CAN YOU ANSWER THESE QUESTIONS? I HAVE NOT DONE A R2CLOUD.TXT THING BEFORE.

This is a first time login. Please setup account:

    Insert your SD card into the card reader
IS THIS A CARD READER ON A DIFFERENT COMPUTER OR IS THIS THE CARD INSTALLED IN A RUNNING PI?
    Open SD card contents and create the file r2cloud.txt there
WHAT FOLDER DOES THIS GO IN?  BOOT?  PI?
    Put some keyword into this file on a single line
SO 9 NUMBERS ARE OK?
    Save the file and open this page once again
AM I SAVING THE FILE IN A CARD READER OR IN THE PI?  AGAIN WHAT FOLDER?
    Enter the keyword, new email and password
IS THIS THE SAME KEYWORD THAT I PUT IN THE R2CLOUD.TXT FILE?

TY

Jeff

jeffreypkelly, 2019-10-25:
Cancel - must have had an extra space at end of keyword - all good.

dernasherbrezon, 2019-10-25:
Better run 2 pi. This will organically double the CPU power + 2 schedulers can track 2 satellites concurrently.

closed 2019-10-25 by dernasherbrezon