dernasherbrezon/r2cloud #58: I am connected and running but not seeing any files saved.

jeffreypkelly, 2019-10-26:
Any suggestions?  I have an rtlsdr feed by a 137 mhz QFH antenna.
Where are files stored locally? How can I trouble shoot?

[image]

dernasherbrezon, 2019-10-26:
Hi! You can click on observation and check spectogram. Also, logs are available in journald:

> journalctl -u r2cloud -n 200

jeffreypkelly, 2019-10-26:
In the log I am getting the usb_claim_interface error -6 error message.

When I reboot and run rtf_test or a FM radio command:
"rtl_fm -g 50 -f 95.9M -M wfm -s 180k -E deemp |  play -r 180k -t raw -e s -b 16 -c 1 -V1 - lowpass 16k” 
everything works well.

Do I have a permissions issue?

Jeff

dernasherbrezon, 2019-10-26:
Maybe. Can you:
1) Check if no other software is using RTL-SDR
2) Check if kernel driver loaded first: lsmod |grep rtl ( See some remediation here https://stackoverflow.com/questions/43879570/usb-claim-interface-error-6-on-rtl-sdr)

jeffreypkelly, 2019-10-26:
> 
> Maybe. Can you:
> 
> Check if no other software is using RTL-SDR
I am using your image and no other software installed.
> Check if kernel driver loaded first: lsmod |grep rtl ( See some remediation here https://stackoverflow.com/questions/43879570/usb-claim-interface-error-6-on-rtl-sdr <https://stackoverflow.com/questions/43879570/usb-claim-interface-error-6-on-rtl-sdr>)
No results with this grep

I have the modprobe.d blacklist files in.

It is funny that I can run rtl_test and a rtl_fm command and not have an error.

OK I’ll mess around.  Is there a way to manually start a recording for r2cloud so I can test manually?

Jeff

dernasherbrezon, 2019-10-26:
There are 2 options:
1) Check the logs for the actual command. It should output something like: ```./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f ...```. You can run this command manually
2) REST API. Authenticate using REST API and execute immediate observation. See https://github.com/dernasherbrezon/r2cloud/blob/master/src/test/java/ru/r2cloud/it/ObservationTest.java for code example.

jeffreypkelly, 2019-10-26:
OK I got into HTOP and killed the pid for rtf that was running and low and behold this shows up from NOAA 19:
So let me monitor this and see if somehow the rtl_fm is not shutting down and thus preventing a restart on the next pass.

TY!

Jeff

jeffreypkelly, 2019-10-27:
Things seem to be working so you can close, delete this.
TY!

Jeff

jeffreypkelly, 2019-10-27:
Can I find my api key on the sd card?

Jeff

jeffreypkelly, 2019-10-27:
Cancel, created account.

Jeff

closed 2019-10-27 by jeffreypkelly