dernasherbrezon/r2cloud #59: Satnogs Integration

jeffreypkelly, 2019-10-27:
Besides enabling Satnogs in r2server, do I have to do configure else?
Do I need an account on Satnogs or add the satnogs api to your server?
If not how does the data show up in Satnogs?  Under name r2cloud? or my name?

dernasherbrezon, 2019-10-27:
You can configure your callsign in r2server and it will upload telemetry using SIDS protocol. No registration at SATNOGS required. You can query SATNOGS later on:

https://db.satnogs.org/api/telemetry/?satellite=43803&app_source=&observer=M7RED-LO33sr&transmitter=&start=&end=

dernasherbrezon, 2019-10-30:
They accept data only via sids protocol and this protocol supports only telemetry. So no complex data like weather images :(

You can try turn on any of funcube satellites.

dernasherbrezon, 2019-10-30:
If you know any other project that accepts NOAA images, then I can write integration with it

dernasherbrezon, 2019-10-30:
Yes, I can make a setting to expose private satellite data straight from r2server.

dernasherbrezon, 2019-10-31:
Raised https://github.com/dernasherbrezon/r2cloud/issues/62

closed 2019-10-31 by dernasherbrezon

mash55, 2019-11-03:
Yeah, I am struggling with the syntax to query the satnogs DB as well. @jeffreypkelly... did you figure out how to query your observations?

dernasherbrezon, 2019-11-03:
For some reason Satnogs adds some random number to the "callsign".

mash55, 2019-11-03:
for each callsign or for each observation? How do I find this random number?

mash55, 2019-11-03:
checked the satnogs DB statstic page and found my callsign followed by the grid square.
satnogs DB also states that the last observation was received on 2019-11-03 17:35:56
Checking my observations, this is most likely the latest NOAA 18 observation.
I used your example from above, replacing your callsign/grid with mine and used the NOAA18 Norad ID to query the satnogs DB - no joy

https://db.satnogs.org/api/telemetry/?satellite=28654&app_source=&observer=K0MOS-DN70ic&transmitter=&start=&end=

I have NOAA 18 observations with data. What am I doing wrong?

dernasherbrezon, 2019-11-03:
NOAA is not being upload there.  https://github.com/dernasherbrezon/r2cloud/issues/59#issuecomment-547986168

mash55, 2019-11-03:
whoops, I read that but apparently brain did not apply info :(
Now it works as described. Would it be possible to add satellite name and Norad ID to the observation (when clicking on the date/time to see META info etc. Having the Name and Norad-ID available on this screen would be helpful - happy to open a new issue.
Thanks

dernasherbrezon, 2019-11-03:
Raised https://github.com/dernasherbrezon/r2cloud/issues/66