dernasherbrezon/r2cloud #60: 502 r2cloud is not running

mash55, 2019-10-27:
Installed from provided image. Acessing the UI via URL generates error message: 502
r2cloud is not running

Any help would be appreciated

dernasherbrezon, 2019-10-27:
Can you login into raspberry pi and get the logs using command:

>  journalctl -u r2cloud -n 200

mash55, 2019-10-27:
Here it is:

Oct 25 20:50:59 raspberrypi systemd[1]: Started R2Cloud Service.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=killed, status=4/ILL
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Unit entered failed state.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Failed with result 'signal'.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Service hold-off time over, scheduling restart.
Oct 25 20:51:00 raspberrypi systemd[1]: Stopped R2Cloud Service.
Oct 25 20:51:00 raspberrypi systemd[1]: Started R2Cloud Service.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=killed, status=4/ILL
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Unit entered failed state.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Failed with result 'signal'.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Service hold-off time over, scheduling restart.
Oct 25 20:51:00 raspberrypi systemd[1]: Stopped R2Cloud Service.
Oct 25 20:51:00 raspberrypi systemd[1]: Started R2Cloud Service.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=killed, status=4/ILL
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Unit entered failed state.
Oct 25 20:51:00 raspberrypi systemd[1]: r2cloud.service: Failed with result 'signal'.
Oct 25 20:51:01 raspberrypi systemd[1]: r2cloud.service: Service hold-off time over, scheduling restart.
Oct 25 20:51:01 raspberrypi systemd[1]: Stopped R2Cloud Service.
Oct 25 20:51:01 raspberrypi systemd[1]: Started R2Cloud Service.
Oct 25 20:51:01 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=killed, status=4/ILL
Oct 25 20:51:01 raspberrypi systemd[1]: r2cloud.service: Unit entered failed state.
Oct 25 20:51:01 raspberrypi systemd[1]: r2cloud.service: Failed with result 'signal'.
Oct 25 20:51:01 raspberrypi systemd[1]: r2cloud.service: Service hold-off time over, scheduling restart.
Oct 25 20:51:01 raspberrypi systemd[1]: Stopped R2Cloud Service.
Oct 25 20:51:02 raspberrypi systemd[1]: Started R2Cloud Service.
Oct 25 20:51:02 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=killed, status=4/ILL
Oct 25 20:51:02 raspberrypi systemd[1]: r2cloud.service: Unit entered failed state.
Oct 25 20:51:02 raspberrypi systemd[1]: r2cloud.service: Failed with result 'signal'.
Oct 25 20:51:02 raspberrypi systemd[1]: r2cloud.service: Service hold-off time over, scheduling restart.
Oct 25 20:51:02 raspberrypi systemd[1]: Stopped R2Cloud Service.
Oct 25 20:51:02 raspberrypi systemd[1]: r2cloud.service: Start request repeated too quickly.
Oct 25 20:51:02 raspberrypi systemd[1]: Failed to start R2Cloud Service.

mash55, 2019-10-27:
Please disregard.... looks like a PI Zero is the issue, dug up a PI3 and it is now working.

closed 2019-10-27 by dernasherbrezon