dernasherbrezon/r2cloud #61: Several observations but no data and nothing on r2server.ru

sysrun, 2019-10-31:
Hi,
i installed r2cloud via apt on my raspberry. It recorded several passes (i can see observations in my local webiterface) and also see files in my /tmp folder:
-rw-r--r--  1 pi   pi   137640592 Oct 31 13:15 39430-1572523511203.raw.gz
-rw-r--r--  1 pi   pi   132630117 Oct 31 14:53 39430-1572529385237.raw.gz
-rw-r--r--  1 pi   pi   120601551 Oct 31 12:35 41460-1572521198485.raw.gz
-rw-r--r--  1 pi   pi   105949818 Oct 31 14:09 41460-1572526888503.raw.gz
-rw-r--r--  1 pi   pi    93033797 Oct 31 12:09 43743-1572519742122.raw.gz

1.) When i click in the "Spectorgram" tab and then "generate" i get a 404 from the api (Unable to find .wav)

2.) I connected to r2server.ru and see the metrics for my basestation - but no observations are reported.

Any help?

sysrun, 2019-10-31:
Log:

> root@astropi:/home# journalctl -u r2cloud -n 400
-- Logs begin at Thu 2019-10-31 14:17:01 CET, end at Thu 2019-10-31 14:58:42 CET. --
Oct 31 14:43:05 astropi java[561]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437268662 -s 240000 -g 45 -p 0 -o /tmp/39430-1572529385237.raw.gz [ru.r2cloud.util.ProcessFactory crea
te]
Oct 31 14:43:05 astropi java[561]: Found 1 device(s):
Oct 31 14:43:05 astropi java[561]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Oct 31 14:43:05 astropi java[561]: Using device 0: Generic RTL2832U OEM
Oct 31 14:43:05 astropi java[561]: Found Rafael Micro R820T tuner
Oct 31 14:43:05 astropi java[561]: [R82XX] PLL not locked!
Oct 31 14:43:05 astropi java[561]: Sampling at 240000 S/s.
Oct 31 14:43:05 astropi java[561]: Tuned to 437268662 Hz.
Oct 31 14:43:05 astropi java[561]: Tuner gain set to 44.50 dB.
Oct 31 14:43:05 astropi java[561]: Reading samples in async mode...
Oct 31 14:43:05 astropi java[561]: Allocating 15 zero-copy buffers
Oct 31 14:51:09 astropi java[561]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Oct 31 14:51:09 astropi java[561]: stopping: rtl-status [ru.r2cloud.util.Util shutdown]
Oct 31 14:52:20 astropi java[561]: data file not found [ru.r2cloud.web.api.observation.ObservationSpectrogram doPost]
Oct 31 14:53:15 astropi java[561]: stopping: rtl_sdr for 1572529385237 [ru.r2cloud.util.Util shutdown]
Oct 31 14:53:15 astropi java[561]: Signal caught, exiting!
Oct 31 14:53:15 astropi java[561]: [1572529385237] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Oct 31 14:53:15 astropi java[561]: User cancel, exiting...
Oct 31 14:53:15 astropi java[561]: scheduled next pass for 39430. start: Fri Nov 01 01:35:43 UTC 2019 end: Fri Nov 01 01:45:02 UTC 2019 [ru.r2cloud.satellite.Scheduler schedule]
Oct 31 14:53:15 astropi java[561]: cancelling previous: 1572529385237 [ru.r2cloud.satellite.Schedule add]
Oct 31 14:57:54 astropi java[561]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Oct 31 14:57:54 astropi java[561]: stopping: rtl-status [ru.r2cloud.util.Util shutdown]
Oct 31 14:58:30 astropi java[561]: data file not found [ru.r2cloud.web.api.observation.ObservationSpectrogram doPost]

dernasherbrezon, 2019-10-31:
Have configured base station in r2server.ru? It needs to be configured there and in the r2cloud. See the guide: https://github.com/dernasherbrezon/r2cloud/wiki/r2server

sysrun, 2019-10-31:
Yes i did. I am already getting the metrics from my installation (like load average, ram etc.)

dernasherbrezon, 2019-10-31:
Several things here:
1) .raw.gz should be in /tmp directory only for a short period of time
2) they might not be moved to /home/pi/r2cloud/data/ folder. Can you check disc space on that partition?

Can you post relevant logs for this observation using the following command:

> journalctl -u r2cloud -n 500 |grep 1572529385237

sysrun, 2019-10-31:
1. Plenty of space left (+40Gig / SSD)
2. The data folder only contains "rrd" &"satellites"

Got a new recording in tmp -> 39444-1572539101255.raw.gz

Logs:

```
root@astropi:~# journalctl -u r2cloud -n 500 |grep 1572539101255
Oct 31 17:25:01 astropi java[560]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 145948267 -s 240000 -g 45 -p 0 -o /tmp/39444-1572539101255.raw.gz [ru.r2cloud.util.ProcessFactory create]
Oct 31 17:34:23 astropi java[560]: stopping: rtl_sdr for 1572539101255 [ru.r2cloud.util.Util shutdown]
Oct 31 17:34:23 astropi java[560]: [1572539101255] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Oct 31 17:34:23 astropi java[560]: cancelling previous: 1572539101255 [ru.r2cloud.satellite.Schedule add]
root@astropi:~#
```

sysrun, 2019-10-31:
Maybe unrelated: when i save the cofiguraten in the web interface i get a 500:

`Oct 31 17:43:17 astropi java[560]: unable to handle request
Oct 31 17:43:17 astropi java[560]: java.lang.IllegalArgumentException: java.nio.file.AtomicMoveNotSupportedException: /tmp/user.properties.tmp -> /home/pi/.r2cloud: Invalid cross-device link
Oct 31 17:43:17 astropi java[560]:         at ru.r2cloud.util.Configuration.update(Configuration.java:100)
Oct 31 17:43:17 astropi java[560]:         at ru.r2cloud.web.api.configuration.R2CloudSave.doPost(R2CloudSave.java:51)
Oct 31 17:43:17 astropi java[560]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:86)
Oct 31 17:43:17 astropi java[560]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Oct 31 17:43:17 astropi java[560]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Oct 31 17:43:17 astropi java[560]:         at java.base/java.lang.Thread.run(Thread.java:834)
Oct 31 17:43:17 astropi java[560]: Caused by: java.nio.file.AtomicMoveNotSupportedException: /tmp/user.properties.tmp -> /home/pi/.r2cloud: Invalid cross-device link
Oct 31 17:43:17 astropi java[560]:         at java.base/sun.nio.fs.UnixCopyFile.move(UnixCopyFile.java:415)
Oct 31 17:43:17 astropi java[560]:         at java.base/sun.nio.fs.UnixFileSystemProvider.move(UnixFileSystemProvider.java:263)
Oct 31 17:43:17 astropi java[560]:         at java.base/java.nio.file.Files.move(Files.java:1421)
Oct 31 17:43:17 astropi java[560]:         at ru.r2cloud.util.Configuration.update(Configuration.java:98)
Oct 31 17:43:17 astropi java[560]:         ... 5 more
Oct 31 17:43:17 astropi java[560]:  [ru.r2cloud.web.WebServer serve]`

So i copied the /tmp/user.properties.tmp to /home/pi/.r2cloud manualy

dernasherbrezon, 2019-10-31:
Actually this is related. Found this helpful answer:
https://www.raspberrypi.org/forums/viewtopic.php?t=71837

Looks like /tmp and /home/pi are on the different devices. Currently I'm doing atomic rename of file. But since src and dst are on the different file systems, atomic operation impossible.

I will put better logging for such case and think on how this could be improved. This is not trivial, as I currently rely on atomic operation: file cannot be corrupted and another process still continue to write to the file handle.

sysrun, 2019-10-31:
I see

Is it safe to use "/home/pi/tmp" then?

dernasherbrezon, 2019-10-31:
Yes, it is.

dernasherbrezon, 2019-11-07:
Wrote steps to mitigate the issue in the troubleshooting guide

closed 2019-11-07 by dernasherbrezon