dernasherbrezon/r2cloud #63: Delete observations

chris-y, 2019-10-31:
Please can we have an option to delete observations locally, either manually or on a schedule (or maybe as soon as they are synced to the server?).

In the meantime is there an easy way to do this from the command line?

mash55, 2019-10-31:
I second that

dernasherbrezon, 2019-11-01:
I can implement that, but why do you want to delete them? They will be automatically deleted based on the retention period. By default it is:

> scheduler.data.retention.count=2

I.e. it will keep last 2 observations.

chris-y, 2019-11-01:
In my case I'm running out of space (my fault for using a too small SD card). I'm going to switch to a bigger SD card, but then I thought I'd just get the same problem again.  If there's already a retention period then that solves that issue.  However, it would still be nice to be able to clear out old readings manually, eg. if there's a satellite you've stopped monitoring the old observations will be there indefinitely otherwise.

dernasherbrezon, 2019-11-01:
Ah yes. Removing unsupported/decayed satellites from SD card makes sense. For now I suggest either:

1. putting retention = 1 and uploading everything into r2server. Or
2. Disabling unwanted satellites. You can check spectogram from any of their observation and see if the signal is strong enough.

mash55, 2019-11-01:
I was unaware of the retention period. If there is documentation on how everything works, please let me know. 

So with the current retention period, data will disappear from the r2cloud server? 

How about an option to download an observation (Metadata/Data/Spectrogram/Remarks (See #64) ) so we can keep 'important' observation locally?

dernasherbrezon, 2019-11-01:
The data won't disappear from r2server. It will be deleted from r2cloud only. 

With "important" observations it would by really hard to calculate space on SD card. On the one hand, they should be deleted because of disk space, on the other keep them, because they important. Maybe it is better to have them on r2server where disk space is not an issue?

mash55, 2019-11-01:
Well, that works for me too. If I can add notes/remarks to observations and they will be transferred to the r2cloud and kept on r2cloud it will be possible to compare (to a certain extend) observations made with different HW (antenna/LNA/diplexres/location (see #64 )

dernasherbrezon, 2020-10-05:
I added a separate retention period for meta data. So it is possible to keep meta data for longer period and cleanup raw data (heavyweight) earlier.

closed 2020-10-05 by dernasherbrezon