dernasherbrezon/r2cloud #67: Selecting Meteor M2 Evening Data Channel

jeffreypkelly, 2019-11-04:
It seems in the NOAA images you select the visual channel in daylight and show both infrared channels in the evening.

With Meteor M2 - it seems you combine the visual Channels 1 & 2 to create a false color image, and keep that in the evening, which shows a no sunlight black image. 
Is there a way to select the Infrared channel in the evening to show?

I realize this can change as the mode of the satellite changes from all visual channels to the current mode two visual and an infrared, so it would nice to be configurable if possible.

dernasherbrezon, 2019-11-05:
The raw data is there, it just do not rendered. The question is, what would you do with images next?

I'm planning to create a service, which aggregates all Meteor M2/2-2 images and lay them on google maps. For this use case, pre-rendered images are useless, as I need raw data with admin packets for each row.

errolt, 2020-02-24:
How can I reprocess the raw data to get all images uncombined?

dernasherbrezon, 2020-02-24:
Hmm Good question. Looks like raw data for Meteor can be downloaded via API only from r2server.

If you have access to raspberry pi, then all raw data stored on the disk. It just do not exposed via API. Raised #77 to fix that.

You can process raw data using the code below:

```java
		try (LRPTInputStream is = new LRPTInputStream(LRPTInputStreamTest.class.getClassLoader().getResourceAsStream("data.bin"))) {
			while(is.hasNext()) {
				Vcdu vcdu = is.next();
			}
		}
```

Vcdu is the transfer frame from LRPT specification. Sample code for joining Vcdu together and extracting Packets is in ```MeteorImage``` class.

dernasherbrezon, 2023-05-21:
LRPT was turned off on METEOR. Use the following program to generate geotiff with all available channels: https://github.com/dernasherbrezon/georef-lrpt

closed 2023-05-21 by dernasherbrezon