dernasherbrezon/r2cloud #68: Accepting self signed certificate

raspimeteo, 2019-11-07:
Just booted this morning my pi with the r2cloud SD. Besides the trouble that the network interface didn't get an DHCP address I now face the next hurdle.

I exported the certificate with development tools of Chrome and imported them manually into Chrome but still can't open the website to start configuring.

Simple instruction would be helpful, accept self signed certificate is not really helping. 

John

raspimeteo, 2019-11-07:
Meanwhile passed that hurdle too. Somehow in IE I only received a login prompt without the key field.
Imported certificate again in Chrome and it worked.

closed 2019-11-07 by raspimeteo

dernasherbrezon, 2019-11-07:
Yes, browsers tend to change the way they handle self signed certificates quite frequently. Also different browsers have different policy around them. For example, it is super easy to accept self-signed certificate in Firefox, but in Chrome it is challenging.