dernasherbrezon/r2cloud #69: Session time from local r2cloud is too short

DE8MSH, 2019-11-23:
Hey.

The session time from the local r2cloud is too short. After 2-3 minutes I have to reenter the name and pw. This is too short for maken any changes, scheduling etc. p. p.

73

dernasherbrezon, 2019-11-23:
Yes, makes sense. Increased to 30 minutes

closed 2019-11-23 by dernasherbrezon

DE8MSH, 2019-12-08:
You can reopen this issue. I talked about LOCAL session time. Still only 1-3 minutes.

dernasherbrezon, 2019-12-08:
Can you check parameter ```server.session.timeout.millis``` in the /home/pi/r2cloud/etc/config-prod.properties ?

DE8MSH, 2019-12-08:
360000. So 6 minutes. I'll set it up to a longer timeout time. Thanks.

dernasherbrezon, 2019-12-08:
Yes, currently it is:

#30 minutes
server.session.timeout.millis=1800000

You can simply turn on automatic updates to get all new stuff daily.

DE8MSH, 2019-12-08:
Thanks. Now you can close this issue.

But the other issue (cannot reset passwort) is still open and most important!!! Help me with it, please. Thank you.