dernasherbrezon/r2cloud #72: Password reset not working. Not localy, not on .ru.

DE8MSH, 2019-12-08:
Hey.

I cannot reset my password. Not localy: I set a r2could.txt in / on 1st partition. Does NOT work. It says everytime "Invalid login or password" after setting up a new pw with r2cliut.txt, e-mail address  and NEW pw in a loop!!!

AND: on https://r2server.ru I reseted my pw, too. Result: Nothing happens. No new info about setting up a new pw for online page.

What's next???

dernasherbrezon, 2019-12-08:
These are 2 independent issues.

1. Where did you put r2cloud.txt? It should be in the ```/boot``` partition rather than ```/```.
2. It was the issue on the server side. Can you please try again?

DE8MSH, 2019-12-08:
1. in /boot/
2. nope. locally it jumps to login scren again. and it's looping all the time between login screen aka password reset and login screen. (locally). login>password reset>restore>setup>login>password reset>restore>setup>etc. p. p.

dernasherbrezon, 2019-12-08:
There are 2 components:

1. r2cloud. Which is the server running on your raspberry pi. It uses file from ```/boot/r2cloud.txt``` to setup initial authentication. Can you check the log? This guide can help https://github.com/dernasherbrezon/r2cloud/wiki/Troubleshooting-guide
2. r2server. This is server-side component running on the internet. It has separate password (not the one you configured for r2cloud!) and separate forget/restore procedure via email. I'm seeing you requested password reset there. Have you received an email? I'm no longer seeing any issues in r2server.ru. Your email server looks like non-public. Can you check its logs?

DE8MSH, 2019-12-08:
2. r2server: 

solved. I sends me an email. I changed it a few minutes ago.

1. r2cloud: 

Dec 08 16:50:23 raspberrypi java[604]: reset password for: xxxxxxxx@xxxxx.xx [ru.r2cloud.web.Authenticator resetPassword]
Dec 08 16:50:33 raspberrypi java[604]: unable to save: {keyword=Invalid keyword} [ru.r2cloud.web.api.setup.Setup doPost]

dernasherbrezon, 2019-12-08:
Good! 

Let's solve r2cloud issue now. According to the [source code](https://github.com/dernasherbrezon/r2cloud/blob/master/src/main/java/ru/r2cloud/web/api/setup/Setup.java#L66): file is there, it has right permissions, just keyword doesn't match.

Can you check if keyword is non blank, non empty string? Try lowercase, copy/paste it from file to the input during configuration?

DE8MSH, 2019-12-08:
Could it be a problem /w UNIX/Windows editors? I set the keyword with nano. If I enter a ONELINER and save it WITHOUT [enter] afterwards (STRG-X) , I'm getting a ONELINER plus a new line???!!!

DE8MSH, 2019-12-08:
Log is:

Dec 08 17:04:48 raspberrypi java[604]: reset password for: xxxxxxxx@xxx.xx [ru.r2cloud.web.Authenticator resetPassword]

But: If I login with new pw:

Invalid login or password

dernasherbrezon, 2019-12-08:
Have you tried lowercase, copy/paste keyword? 

The logs are ok. Here is what happening:

1. ```ru.r2cloud.web.Authenticator resetPassword```. Will reset login and password. No auth possible until you setup new password using keyword from r2cloud.txt
2. ```unable to save: {keyword=Invalid keyword} ``` - keyword provided didn't match the one in the r2cloud.txt. r2cloud still not configured after that step. So no auth possible. You can't login until you specify correct keyword.

DE8MSH, 2019-12-09:
Hi.

Forget about 2. It was a check which keyword ist the right one. 

1. IT DOES NOT WORK. This is what happens all the  time:

https://imgur.com/a/FSYEBCQ

Key is correct.

dernasherbrezon, 2019-12-09:
The password is stored locally on raspberrypi: ```/home/pi/.r2cloud```. The password itself is salted. So you can verify manually:

```bash
echo -n "password{salt}" | openssl dgst -sha256
```

Where password - "your password", salt - "server.salt". Then compare result with the property value "server.password".

DE8MSH, 2019-12-09:
Thank you @dernasherbrezon.

I'll give it a try t'night.

l8er.

DE8MSH, 2019-12-09:
I does not write a or the new password. The salted pw stays in the .r2cloud file.

dernasherbrezon, 2019-12-09:
1. Can you check the log for any errors?
2. Can you check file permissions /home/pi/.r2cloud file ?
3. Can you check if "server.password" is there? Does it change if you setup different password?
4. Can you clean browser cache?
5. Can you delete /home/pi/.r2cloud file and re-configure again?

DE8MSH, 2019-12-09:
2. pi/pi
3. it does not change
4. i did
5. if i delete .r2cloud: how to enable initial start again?
6. https://raspberrypi.local/api/v1/accessToken 401 (Unauthorized) <<<< means?

dernasherbrezon, 2019-12-09:
3. Ok. This is happening only when you specify invalid username during "restore". Can you check "server.login" property from /home/pi/.r2cloud?
5. Just open browser as normal. It will re-init initial setup
6. This is ok if you're not authenticated.

DE8MSH, 2019-12-09:
4. I deleted .r2cloud

I called raspberrypi.local again. No initial setup.

dernasherbrezon, 2019-12-09:
You have to restart r2cloud or raspberry

DE8MSH, 2019-12-10:
Wow. This was a hard birth. I deleted .r2cloud, rebooted the Pi and now I have to initial setup screen again.

BUT: Resetting pw did not work at all. And I don't know why yet.

closed 2020-04-24 by dernasherbrezon