dernasherbrezon/r2cloud #73: NOAA map overlay

chris-y, 2019-12-16:
Is it possible to add a map overlay to the NOAA satellite images?
I believe the wxmap tool can generate a map which is then fed into wxtoimg to overlay it.  wxmap is already present on the r2cloud image.

dernasherbrezon, 2019-12-18:
Why do you want this overlay? 

Instead of adding custom data onto image I suggest converting raw chanenls into GeoTIFF format and process whatever you like. I raised jira for decoder similar to wxtoimg. https://github.com/martinber/noaa-apt/issues/6

So instead of adding map overlay to image, let's add image to map! I.e. google map.

chris-y, 2019-12-18:
Basically so I can tell where the clouds are!  Adding the image to a map would certainly work too.