dernasherbrezon/r2cloud #79: sorry about I can not link ur deb

zhangyulu8023, 2020-03-28:
建议安装：
  libsox-fmt-all fcgiwrap nginx-doc ssl-cert
下列【新】软件包将被安装：
  libhyperic-sigar-java libnginx-mod-http-echo libopencore-amrnb0 libopencore-amrwb0
  libsox-fmt-alsa libsox-fmt-base libsox2 nginx-common nginx-light r2cloud r2cloud-jdk
  r2cloud-ui sox wxtoimg
下列软件包将被升级：
  librtlsdr0 rtl-sdr
升级了 2 个软件包，新安装了 14 个软件包，要卸载 0 个软件包，有 1 个软件包未被升级。
需要下载 54.5 MB 的归档。
解压缩后会消耗 24.6 MB 的额外空间。
您希望继续执行吗？ [Y/n] y
获取:1 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libopencore-amrnb0 armhf 0.1.3-2.1 [80.4 kB]
获取:2 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libopencore-amrwb0 armhf 0.1.3-2.1 [42.0 kB]
获取:3 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libhyperic-sigar-java armhf 1.6.4+dfsg-2.1 [379 kB]
获取:4 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf nginx-common all 1.10.3-1+deb9u3 [105 kB]
获取:5 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libnginx-mod-http-echo armhf 1.10.3-1+deb9u3 [95.4 kB]
获取:6 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf nginx-light armhf 1.10.3-1+deb9u3 [373 kB]
获取:7 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libsox2 armhf 14.4.1-5+deb9u2 [230 kB]
获取:8 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libsox-fmt-alsa armhf 14.4.1-5+deb9u2 [47.1 kB]
获取:9 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf libsox-fmt-base armhf 14.4.1-5+deb9u2 [64.7 kB]
获取:10 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf sox armhf 14.4.1-5+deb9u2 [135 kB]
错误:11 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud-jdk armhf 20191226160628     
  无法连接上 s3.amazonaws.com:80 (52.216.229.125)，连接超时 [IP: 52.216.229.125 80]
错误:12 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud-ui armhf 1583109042
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:13 http://s3.amazonaws.com/r2cloud r2cloud/main armhf rtl-sdr armhf 0.6git
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:14 http://s3.amazonaws.com/r2cloud r2cloud/main armhf librtlsdr0 armhf 0.6git
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:15 http://s3.amazonaws.com/r2cloud r2cloud/main armhf wxtoimg armhf 2.11.2
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:16 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud armhf 20200327202312
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
已下载 1,552 kB，耗时 47秒 (33.2 kB/s)              
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/r/r2cloud-jdk/r2cloud-jdk_20191226160628_armhf.deb  无法连接上 s3.amazonaws.com:80 (52.216.229.125)，连接超时 [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/r2cloud/r/r2/r2cloud-ui-1583109042.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/r/rtl-sdr/rtl-sdr_0.6git_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/l/librtlsdr0/librtlsdr0_0.6git_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/w/wxtoimg/wxtoimg_2.11.2_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/r2cloud/r/r2/r2cloud-20200327202312.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 有几个软件包无法下载，要不运行 apt-get update 或者加上 --fix-missing 的选项再试试？
root@raspberrypi:/home/pi# sudo apt-get install r2cloud
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
下列软件包是自动安装的并且现在不需要了：
  coinor-libcbc3 coinor-libcgl1 coinor-libclp1 coinor-libcoinmp1v5 coinor-libcoinutils3v5
  coinor-libosi1v5 fonts-noto-core fonts-noto-ui-core fonts-sil-gentium fonts-sil-gentium-basic
  gstreamer1.0-gtk3 libbsh-java libcdr-0.1-1 libcolamd2 libfreehand-0.1-1 libmspub-0.1-1
  libpagemaker-0.0-0 libqxp-0.0-0 libreoffice-avmedia-backend-gstreamer libreoffice-calc
  libreoffice-draw libreoffice-gtk3 libreoffice-impress libreoffice-librelogo
  libreoffice-nlpsolver libreoffice-script-provider-bsh libreoffice-script-provider-js
  libreoffice-script-provider-python libreoffice-wiki-publisher libsuitesparseconfig5
  libvisio-0.1-1 libzmf-0.0-0 lp-solve
使用'sudo apt autoremove'来卸载它(它们)。
将会同时安装下列软件：
  libhyperic-sigar-java libnginx-mod-http-echo libopencore-amrnb0 libopencore-amrwb0 librtlsdr0
  libsox-fmt-alsa libsox-fmt-base libsox2 nginx-common nginx-light r2cloud-jdk r2cloud-ui
  rtl-sdr sox wxtoimg
建议安装：
  libsox-fmt-all fcgiwrap nginx-doc ssl-cert
下列【新】软件包将被安装：
  libhyperic-sigar-java libnginx-mod-http-echo libopencore-amrnb0 libopencore-amrwb0
  libsox-fmt-alsa libsox-fmt-base libsox2 nginx-common nginx-light r2cloud r2cloud-jdk
  r2cloud-ui sox wxtoimg
下列软件包将被升级：
  librtlsdr0 rtl-sdr
升级了 2 个软件包，新安装了 14 个软件包，要卸载 0 个软件包，有 1 个软件包未被升级。
需要下载 52.9 MB/54.5 MB 的归档。
解压缩后会消耗 24.6 MB 的额外空间。
您希望继续执行吗？ [Y/n] y
错误:1 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud-jdk armhf 20191226160628      
  无法连接上 s3.amazonaws.com:80 (52.216.229.125)，连接超时 [IP: 52.216.229.125 80]
错误:2 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud-ui armhf 1583109042
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:3 http://s3.amazonaws.com/r2cloud r2cloud/main armhf rtl-sdr armhf 0.6git
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:4 http://s3.amazonaws.com/r2cloud r2cloud/main armhf librtlsdr0 armhf 0.6git
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:5 http://s3.amazonaws.com/r2cloud r2cloud/main armhf wxtoimg armhf 2.11.2
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
错误:6 http://s3.amazonaws.com/r2cloud r2cloud/main armhf r2cloud armhf 20200327202312
  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/r/r2cloud-jdk/r2cloud-jdk_20191226160628_armhf.deb  无法连接上 s3.amazonaws.com:80 (52.216.229.125)，连接超时 [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/r2cloud/r/r2/r2cloud-ui-1583109042.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/r/rtl-sdr/rtl-sdr_0.6git_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/l/librtlsdr0/librtlsdr0_0.6git_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/w/wxtoimg/wxtoimg_2.11.2_armhf.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/r2cloud/r/r2/r2cloud-20200327202312.deb  不能连接到 s3.amazonaws.com：http： [IP: 52.216.229.125 80]
E: 有几个软件包无法下载，要不运行 apt-get update 或者加上 --fix-missing 的选项再试试？
root@raspberrypi:/home/pi#

dernasherbrezon, 2020-03-28:
Hmm. It looks like it cannot find URL:

> E: 无法下载 http://s3.amazonaws.com/r2cloud/pool/main/r/r2cloud-jdk/r2cloud-jdk_20191226160628_armhf.deb 

Can you to download it manually? 

> curl -v http://s3.amazonaws.com/r2cloud/pool/main/r/r2cloud-jdk/r2cloud-jdk_20191226160628_armhf.deb

Most likely you have connectivity issue on raspberry pi.

zhangyulu8023, 2020-03-28:
Thanks for ur latter

And I can wget it but i have a buzy net,it's only 5kb/s ,more than 4h40min……hmm

Sincerity！

dernasherbrezon, 2020-03-28:
You can increase timeout for apt. For example like this:
https://askubuntu.com/questions/141513/how-to-lower-wait-time-for-repository-updates

Or download these packages manually and install them:

> dpkg -i package.deb

closed 2020-03-28 by dernasherbrezon

zhangyulu8023, 2020-03-28:
73

 
Closed #79.