dernasherbrezon/r2cloud #80: Schedule Tab personalization - Rtl_fm reception options.

iu0azb, 2020-04-07:
Would be useful choose order the schedule by time or satellite name. 
Another useful add would be modify from web the rtl_fm parameters, like agc mode, gain etc.. this to optimize exception easily.

dernasherbrezon, 2020-04-20:
Fixed sort by satellite name and next pass time.

iu0azb, 2020-04-25:
Great! it's now easier to verified incoming sats.

dernasherbrezon, 2020-07-14:
I added server time to the overview page. Most of the issues with observations related to the incorrectly configured server time. raspberry pi by default doesn't have realtime clock. So working without internet connection ( and NTP ) could trigger lots of strange issues. 

[image]

gain and bias-t settings will be handled in #89

closed 2020-07-14 by dernasherbrezon