dernasherbrezon/r2cloud #81: RU2 server dont reset password or not register

Eric738, 2020-04-24:
GE
i try sometime today to receive mail with url to reset password to access server to understand how r2cloud run..may be a little mail probl ? 
thank to check
eric

dernasherbrezon, 2020-04-24:
Just tried to reset password and it worked fine. What email address are you using? Can you try well-known mail provider?

Eric738, 2020-04-25:
GM
i use gmail adress, and try again morning without success. Does the password is the same use when i connect  raspberry in local ? thank

dernasherbrezon, 2020-04-25:
1.  r2server.ru is indépendant from the local raspberry pi. 
2. if you reset password, then you can't reset it multiple times. This can DDOS the service. You can click on the link in the first email. It should work.

Eric738, 2020-04-25:
i understand , its ok I did a bad manipulation and I deleted the spam, following your instructions I used a VPN and logged in with another email address and it's ok..My station is connected..next now ..i try to understand why despite very signal reception no image image of noaa or meteor only 1/4 pic of oscar 92...

dernasherbrezon, 2020-04-25:
Glad it worked! There is known issue with QPSK demodulator I'm trying to fix. Feel free to join the discussion at https://gitter.im/r2cloud/Lobby

As for NOAA, I'm using stock wxtoimg. You can check the command line arguments in the logs:
> journalctl -u r2cloud -n 200

closed 2020-04-25 by dernasherbrezon