dernasherbrezon/r2cloud #83: ISS voice

iu0azb, 2020-04-25:
Since ISS can be predicted just adding it to the watch list, would be great record audio on 145.800mhz when pass is expected. Glad to know your opinion ;)

dernasherbrezon, 2020-04-25:
Yes, we can add ISS voice to the list. But what will you do with it? 

I prefer to spend time on "mining" unique and useful data from space. Some examples:

1. Data from weather satellites. 
2. Data from cubesats - unique scientific data, which will help develop better space technologies.

iu0azb, 2020-04-26:
Indeed thinking better  I fully agree, it would have been just a playful extra

ea5wa, 2020-10-04:
It would be nice to listen to the FM or ISS passes. Is one of the features that I prefer from Satnogs.

We are a lot of Ham Radio Operators here and some of us, enjoy listening to the satellites passes

73's from EA5WA Juan Carlos