dernasherbrezon/r2cloud #84: Some questions ...

Eric738, 2020-04-26:
GM

- in my list i cant see nooa 18 , how add easily this satellite or an other satellite ?
 - no pics received from météor or noaa, but signal from other satellites seems ok and big signal,  should we add processes? where can i find help or a tutorial without bothering you every time :) i'm almost a beginner and it is not easy to understand .., the program is much less deaf than satnog that i tested in the same configuration which does not receive much ... thank you in advance

Eric

dernasherbrezon, 2020-04-26:
Hey,

1. NOAA 18 should be in the list available satellites (see top level menu "Satellites"). Do you have screenshot? 
2. The signal on spectrogram is normalized. So even if you see strong (red or yellow) signal on the spectrogram, r2cloud demodulators might not pick it up. You can leave base station for some time and collect results over couple of days.
3. Another important thing is min and guaranteed elevation. Some satellites are sending weak signals. You can tune these parameters based on your antenna. Current parameters:
> scheduler.elevation.min=8
> scheduler.elevation.guaranteed=20
4. Meteor demodulation takes time. If you see observation and its spectrogram but no data, then just wait for 1 hour or maybe more. It takes time to decode it.
5. Please join https://gitter.im/r2cloud/Lobby , we can help you with adhoc questions there

closed 2020-04-26 by dernasherbrezon

Eric738, 2020-04-26:
Hi

oh , sorry i found Noaa 18..only one pass in my page..:)
ok for all infos, i stay one hour on page on meteor to check.. and i join https://gitter.im/r2cloud/lobby. Thank