dernasherbrezon/r2cloud #87: Spectrogram: Internal Server Error

ChadOhman, 2020-06-09:
Hello, just installed r2cloud on a Raspberry Pi 3B+. Had a few NOAA passes but all three passes give me this error when I ask it to generate a spectrogram. What's happening?

[image]

Thanks!

dernasherbrezon, 2020-06-09:
Hey, can you check the logs? 

> journalctl -u r2cloud -n 200

ChadOhman, 2020-06-09:
`
pi@satpi:~ $ journalctl -u r2cloud -n 200
-- Logs begin at Sat 2020-03-21 13:06:14 GMT, end at Tue 2020-06-09 09:35:21 BST
Jun 09 05:47:19 satpi java[538]: Found 1 device(s):
Jun 09 05:47:19 satpi java[538]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 09 05:47:19 satpi java[538]: Using device 0: Generic RTL2832U OEM
Jun 09 05:47:19 satpi java[538]: usb_claim_interface error -6
Jun 09 05:47:19 satpi java[538]: Failed to open rtlsdr device #0.
Jun 09 05:47:19 satpi java[538]: [1591678039368] invalid response code rtl_sdr:
Jun 09 05:47:19 satpi java[538]: scheduled next pass for 43798. start: Tue Jun 0
Jun 09 05:47:19 satpi java[538]: cancelling previous: 1591678039368 [ru.r2cloud.
Jun 09 05:58:02 satpi java[538]: started with arguments: ./etc/rtl_sdr_wrapper.s
Jun 09 05:58:02 satpi java[538]: Found 1 device(s):
Jun 09 05:58:02 satpi java[538]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 09 05:58:02 satpi java[538]: Using device 0: Generic RTL2832U OEM
Jun 09 05:58:02 satpi java[538]: usb_claim_interface error -6
Jun 09 05:58:02 satpi java[538]: Failed to open rtlsdr device #0.
Jun 09 05:58:02 satpi java[538]: [1591678682791] invalid response code rtl_sdr:
Jun 09 05:58:03 satpi java[538]: scheduled next pass for 43814. start: Wed Jun 1
Jun 09 05:58:03 satpi java[538]: cancelling previous: 1591678682791 [ru.r2cloud.
Jun 09 06:14:44 satpi java[538]: started with arguments: ./etc/rtl_sdr_wrapper.s
Jun 09 06:14:44 satpi java[538]: Found 1 device(s):
Jun 09 06:14:44 satpi java[538]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 09 06:14:44 satpi java[538]: Using device 0: Generic RTL2832U OEM
Jun 09 06:14:44 satpi java[538]: usb_claim_interface error -6
Jun 09 06:14:44 satpi java[538]: Failed to open rtlsdr device #0.
Jun 09 06:14:44 satpi java[538]: [1591679684583] invalid response code rtl_sdr:
Jun 09 06:14:45 satpi java[538]: scheduled next pass for 43792. start: Thu Jun 1
Jun 09 06:14:45 satpi java[538]: cancelling previous: 1591679684583 [ru.r2cloud.
Jun 09 06:15:07 satpi java[538]: started with arguments: ./etc/rtl_sdr_wrapper.s
Jun 09 06:15:07 satpi java[538]: Found 1 device(s):
Jun 09 06:15:07 satpi java[538]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 09 06:15:07 satpi java[538]: Using device 0: Generic RTL2832U OEM
Jun 09 06:15:07 satpi java[538]: usb_claim_interface error -6
Jun 09 06:15:07 satpi java[538]: Failed to open rtlsdr device #0.
Jun 09 06:15:07 satpi java[538]: [1591679706972] invalid response code rtl_sdr:
Jun 09 06:15:07 satpi java[538]: scheduled next pass for 43137. start: Tue Jun 0
Jun 09 06:15:07 satpi java[538]: cancelling previous: 1591679706972 [ru.r2cloud.
Jun 09 06:22:09 satpi java[538]: started with arguments: [rtl_test, -t] [ru.r2cl
Jun 09 06:22:09 satpi java[538]: stopping: rtl-status [ru.r2cloud.util.Util shut
Jun 09 06:22:15 satpi java[538]: generating spectogram [ru.r2cloud.SpectogramSer
Jun 09 06:22:15 satpi java[538]: unable to create spectogram
Jun 09 06:22:15 satpi java[538]: java.lang.IllegalArgumentException: not enough
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.jradio.sink.Spectogram.pr
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.SpectogramService.createF
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.SpectogramService.create(
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.web.api.observation.Obser
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.web.WebServer.processPost
Jun 09 06:22:15 satpi java[538]:         at ru.r2cloud.web.WebServer.serve(WebSe
Jun 09 06:22:15 satpi java[538]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.
Jun 09 06:22:15 satpi java[538]:         at fi.iki.elonen.NanoHTTPD$ClientHandle
Jun 09 06:22:15 satpi java[538]:         at java.base/java.lang.Thread.run(Threa
Jun 09 06:22:15 satpi java[538]:  [ru.r2cloud.SpectogramService createFromWav]
Jun 09 06:25:19 satpi java[538]: generating spectogram [ru.r2cloud.SpectogramSer
Jun 09 06:25:19 satpi java[538]: unable to create spectogram
Jun 09 06:25:19 satpi java[538]: java.lang.IllegalArgumentException: not enough
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.jradio.sink.Spectogram.pr
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.SpectogramService.createF
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.SpectogramService.create(
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.web.api.observation.Obser
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.web.WebServer.processPost
Jun 09 06:25:19 satpi java[538]:         at ru.r2cloud.web.WebServer.serve(WebSe
Jun 09 06:25:19 satpi java[538]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.
Jun 09 06:25:19 satpi java[538]:         at fi.iki.elonen.NanoHTTPD$ClientHandle
Jun 09 06:25:19 satpi java[538]:         at java.base/java.lang.Thread.run(Threa
Jun 09 06:25:19 satpi java[538]:  [ru.r2cloud.SpectogramService createFromWav]
Jun 09 06:31:25 satpi java[538]: started with arguments: ./etc/rtl_sdr_wrapper.s
Jun 09 06:31:25 satpi java[538]: Found 1 device(s):
Jun 09 06:31:25 satpi java[538]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 09 06:31:25 satpi java[538]: Using device 0: Generic RTL2832U OEM
[... 110 lines cut ...]
Jun 09 08:58:12 satpi java[538]:         at java.net.http/jdk.internal.net.http.
Jun 09 08:58:12 satpi java[538]:         ... 19 more
Jun 09 08:58:12 satpi java[538]:  [ru.r2cloud.cloud.R2ServerClient lambda$saveMe
Jun 09 09:03:28 satpi java[538]: started with arguments: ./etc/rtl_sdr_wrapper.s
[... cut ...]
Jun 09 09:29:04 satpi java[538]: usb_claim_interface error -6
Jun 09 09:29:04 satpi java[538]: Failed to open rtlsdr device #0.
Jun 09 09:29:04 satpi java[538]: [1591691769943] invalid response code rtl_sdr:
Jun 09 09:29:04 satpi java[538]: scheduled next pass for 43017. start: Wed Jun 1
Jun 09 09:29:04 satpi java[538]: cancelling previous: 1591691769943 [ru.r2cloud.
Jun 09 09:34:59 satpi java[538]: started with arguments: [rtl_test, -t] [ru.r2cl
Jun 09 09:34:59 satpi java[538]: stopping: rtl-status [ru.r2cloud.util.Util shut
lines 179-201/201 (END)
`

dernasherbrezon, 2020-06-09:
The root cause is here:

> Jun 09 06:15:07 satpi java[538]: usb_claim_interface error -6

Something else is using rtl_sdr. You can check this:

> ps aux|grep rtl_sdr

In the meantime I will improve error reporting in r2cloud.

ChadOhman, 2020-06-09:
Interesting. I flashed an SD card with the latest r2cloud image.

```
pi@satpi:~ $ ps aux|grep rtl_sdr
pi         618  0.0  0.1   7676  1760 ?        S    Jun08   0:00 /bin/bash ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 436369147 -s 240000 -g 45 -p 0 -o /tmp/45115-1591650799565.raw.gz
pi         619  0.2  0.2  14880  1932 ?        Sl   Jun08   1:50 rtl_sdr -f 436369147 -s 240000 -p 0 -g 45 -
pi        3830  0.0  0.0   7348   536 pts/0    S+   09:52   0:00 grep --color=auto rtl_sdr
```

dernasherbrezon, 2020-06-09:
Ok. Look like the process hanged. Have you killed java manually? 

Can you do:
1) Kill rtl_sdr and rtl_sdr_wrapper.sh
2) Remove tmp file: rm /tmp/*.raw.gz
3) Restart r2cloud: sudo systemctl restart r2cloud.

ChadOhman, 2020-06-09:
Hi, I ended up doing a full reboot before bed. Turns out that solved it and it seemed to have detected that tmp file and deleted it on its own.