dernasherbrezon/r2cloud #89: How to setup bias-t and gain?

jmceara, 2020-07-11:
Hi.
I'm trying r2cloud for the first time. I'm using pre-build image. How do I set dongle gain and enable bias-t (rtl-sdr v3 dongle)?
I mean, bias-t always be enabled, even after reboot.

I have also tried to set bias-t manually, but doesn't seen to work. I've got this error:

```
pi@raspberrypi:~ $ sudo rtl_biast -b 1
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
rtl_biast: symbol lookup error: rtl_biast: undefined symbol: rtlsdr_set_gpio
```

dernasherbrezon, 2020-07-12:
gain is automatically configured. By default:

> satellites.rtlsdr.gain=45

You can overwrite it in ```~/.r2cloud``` file. Just put the value you need and r2cloud will take it before the default.

I will have a look into bias-t issue.

jmceara, 2020-07-12:
May I have one sugestion?
Add gain and bias control in web interface. I've add manual gain control in ```~/.r2cloud``` file....but after changing schedule (in web interface), Gain configuration is lost. There's nothing related to gain in the file anymore.

dernasherbrezon, 2020-07-14:
Ok. Looks like rtl_biast program is compatible only with librtlsdr by rtl-sdr.com blog. The latest [osmocom](https://github.com/osmocom/rtl-sdr) rtl-sdr has rtl_biast as well. I upgraded rtl-sdr package to use the latest code and rtl_biast program.

dernasherbrezon, 2020-07-14:
As for gain, you have to restart r2cloud after you configured override. So the steps are the following:

1. Change the gain setting override in ~/.r2cloud
2. Restart r2cloud
3. Enable bias-t manually ```sudo rtl_biast -b 1```

Actually it is a good idea to configure bias-t and gain via web interface. Given the increasing demand, I will start working on it.

dernasherbrezon, 2020-07-14:
Requirements:
* configure in the "general"  settings
* store in each observation for reference 
* show gain and bias-t status in the observation stats on the left

dernasherbrezon, 2020-07-15:
Fixed.

closed 2020-07-15 by dernasherbrezon