dernasherbrezon/r2cloud #91: 502 after powercut

zeroby0, 2020-07-24:
I have frequent power cuts, and after a power loss, r2cloud doesn't start when power is back.

I tried manually starting with systemctl, and see a Java error:
```
Jul 24 13:58:39 raspberrypi java[785]: Exception in thread "main" java.lang.NoClassDefFoundError: org/orekit/time/TimeScale
Jul 24 13:58:39 raspberrypi java[785]:         at ru.r2cloud.R2Cloud.<init>(R2Cloud.java:162)
Jul 24 13:58:39 raspberrypi java[785]:         at ru.r2cloud.R2Cloud.main(R2Cloud.java:301)
Jul 24 13:58:39 raspberrypi java[785]: Caused by: java.lang.ClassNotFoundException: org.orekit.time.TimeScale
Jul 24 13:58:39 raspberrypi java[785]:         at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
Jul 24 13:58:39 raspberrypi java[785]:         at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
Jul 24 13:58:39 raspberrypi java[785]:         at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
Jul 24 13:58:39 raspberrypi java[785]:         ... 2 more
Jul 24 13:58:39 raspberrypi systemd[1]: Stopping R2Cloud Service...
Jul 24 13:58:39 raspberrypi systemd[1]: r2cloud.service: Main process exited, code=exited, status=1/FAILURE
Jul 24 13:58:39 raspberrypi systemd[1]: r2cloud.service: Failed with result 'exit-code'.
Jul 24 13:58:39 raspberrypi systemd[1]: Stopped R2Cloud Service.
```

[log.txt](https://github.com/dernasherbrezon/r2cloud/files/4970935/log.txt)

R2cloud doesn't recover even on systemctl restart or a full reboot and is stuck on 502. The same error is seen on reboot as well.

dernasherbrezon, 2020-07-24:
Looks like installation got corrupted somehow. 

Check the list of libraries:

> ls -lh /usr/share/java/r2cloud/ | wc -l
> 31

If its not 31, then try reinstalling r2cloud:

> sudo apt-get remove r2cloud
> sudo apt-get install r2cloud

closed 2020-07-24 by dernasherbrezon

zeroby0, 2020-07-25:
Yes, it has indeed gotten corrupted. I see 23 instead of 31. I have updates disabled, so it's surprising that the directory would get corrupted.

The librtlsdr package has a version mismatch which prevents me from removing and uninstalling. Is there a way to prevent the corruption from happening, other than getting a UPS?

<pre><code>

$ sudo apt-get remove r2cloud
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
<strong>  rtl-sdr : Depends: librtlsdr0 (= 0.6git) but 0.6.4 is to be installed</strong>
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

$ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  rtl-sdr
The following packages will be upgraded:
  rtl-sdr
1 upgraded, 0 newly installed, 0 to remove and 60 not upgraded.
2 not fully installed or removed.
Need to get 0 B/53.6 kB of archives.
After this operation, 10.2 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Reading changelogs... Done
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 40493 files and directories currently installed.)
Preparing to unpack .../rtl-sdr_0.6.4_armhf.deb ...
<strong>Unpacking rtl-sdr (0.6.4) over (0.6git) ...</strong>
dpkg: error processing archive /var/cache/apt/archives/rtl-sdr_0.6.4_armhf.deb (--unpack):
  trying to overwrite '/usr/bin/rtl_biast', which is also in package rtl-biast 1.0
Errors were encountered while processing:
  /var/cache/apt/archives/rtl-sdr_0.6.4_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
</code></pre>

dernasherbrezon, 2020-07-25:
Try removing rtl-biast package. The new rtl-sdr 0.6.4 has rtl_biast embedded.