dernasherbrezon/r2cloud #95: 17/5000 No NOAA pictures

EA4GSX, 2020-09-20:
Hello
I really like the project, it is very interesting but I have several problems. With the same configuration in windows I can decode more frames. Nor can I receive images from the meteorological satellites. Could it be the saturation of the LNA?
Thanks in advance and congratulations on the project
[image]

dernasherbrezon, 2020-09-21:
Hey,

Yes, it could be. Or antenna was not connected to the rtl-sdr. I cannot see anything on the spectrogram.

closed 2020-09-21 by dernasherbrezon

EA4GSX, 2020-09-21:
Hello again
The antenna was connected  with NOAA you don't see a signal but with the rest you can see a lot of the Doppler. This is normal?
[image]

dernasherbrezon, 2020-09-21:
Hi! Yes, spectrogram contains the visualization of raw i/q data (.wav file) before applying any corrections to the data.

EA4GSX, 2020-09-21:
Okay
Thank you very much for your help, finally, why could the problem be with the noaa? with the METEOR if I see a signal in the waterfall

Thank you

dernasherbrezon, 2020-09-21:
> why could the problem be with the noaa? 

Do you see the same picture for all NOAA satellites or just one pass had it?

EA4GSX, 2020-09-22:
Hello
The problem is with all NOAA passes