dernasherbrezon/r2cloud #96: sub-process /usr/bin/dpkg returned an error code (1) in kali linux,,please help me,i've been bathling this for three days now

Angel211, 2020-09-23:
Please pros help me out because I've used all sort of command and nothing yet,currently in tty mode , I've tried purging and removing some interrupted installations to reinstalled them back especially lightdm as my display manager and GNOME/xfce desktop environment yet each time it boots blankblue screen pops up tho the login bottom isn't really working and I can't login anymore,I'm confused and frustrated right ,I'm using a VMware and running Kali 2017 the updated and upgraded already and problem seem to be btwn my desktop environment and dpkg installation kali theme refuse to install properly ,often get dpkg divert:error unknown renamed or something of that sort , E:subprocess user/bin/dpkg returned errors, please review the screenshot and give me heads up

[image]
[image]
[image]

I can only snap using my poor smaller phone, my laptop is a mess owning to this ,please I plead you take time and review the screenshot of my problem, I don't want to reinstalled Kali and start updating and upgrading all over again

dernasherbrezon, 2020-09-24:
Hmm. Looks like unrelated to the r2cloud. 

It might be caused by inconsistent packages. The root cause is in kali-themes:

> unknown option --no-rename

Try googling it or raising issue in their bug tracker. https://gitlab.com/kalilinux/packages/kali-themes/-/issues

closed 2020-09-24 by dernasherbrezon

Angel211, 2020-09-24:
Gitlab refuse to help saying it's not a support channel, what command do I use pls and is it normal to upgrade dpkg to the latest version?