dernasherbrezon/r2cloud #103: ISS SSTV

arunderwood, 2020-12-20:
I wanted to get your opinion on adding support for decoding SSTV transmissions from the ISS. If you're unaware, the ISS transmits SSTV images during discrete periods a number of times per year.

https://www.rtl-sdr.com/sstv-from-the-international-space-station-scheduled-for-dec-24-dec-31/

Reading through the codebase I believe I understand how I would add support for tracking ISS passes and recording the correct frequencies. Adding a new SSTV decoder would take some more discovery on my part.

Is this something you would be interested in implementing, or receiving support in implementing?

dernasherbrezon, 2020-12-21:
It can be tricky:

1. r2cloud assumes all satellites constantly transmitting data. SSTV transmissions are rare. So most of the time ISS SSTV will be disabled. During the event base station operator should login and manually enabled it.
2. SSTV decoder is harder to implement. I'm not sure if jradio has enough blocks to do this.

We're welcome to implement and add support for ISS SSTV, but I would spend time on something more practical :) For example, HRIT/ LRIT protocol from weather satellites or any cubesat: https://db.satnogs.org/satellites/

Anjum9694, 2021-01-12:
Can it be made to record the SSTV transmission as a wav file? I can just download and feed it to a decoder from PC. Since its a rare event I'm sure people don't mind manually decoding it :)

arunderwood, 2021-01-12:
It's nowhere near as polished as r2cloud, but I ended up using a modified version of this script to capture passes to audio files.

https://github.com/davidhoness/sstv_decoder