dernasherbrezon/r2cloud #106: multiple rtl dongles fail on Pi 4

k7hak, 2021-01-18:
pi 4 fails with NESDR SMArt XTR (E4000) and rtl2838
pi 3 b+ both the dongles work fine tried to get more cpu with the 4 and it looks like its a no go both basic stacks with nothing else added. Same on usb3 ports vs usb2 ports.

pi@r2cloud:~ $ rtl_test
No supported devices found.

pi@r2cloud:~ $ rtl_test
Found 1 device(s):
  0:  , , SN:

Using device 0:
Failed to open rtlsdr device #0.

pi@r2cloud:~ $ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.

pi@r2cloud:~ $ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
rtlsdr_write_reg failed with -7
rtlsdr_write_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_read_reg failed with -4
rtlsdr_write_reg failed with -4
rtlsdr_read_reg failed with -4
rtlsdr_write_reg failed with -4
rtlsdr_read_reg failed with -4
rtlsdr_write_reg failed with -4
rtlsdr_read_reg failed with -4
rtlsdr_write_reg failed with -4
No supported tuner found
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
Enabled direct sampling mode, input 1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
Supported gain values (1): 0.0
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
WARNING: Failed to set sample rate.
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_write_reg failed with -4
rtlsdr_write_reg failed with -4

dernasherbrezon, 2021-01-18:
Doesn't look like r2cloud related. r2cloud supports only single rtl-sdr dongle. 4 CPU cores used for:

1. Read the data from the stick using rtl_sdr program
2. Demodulate/decode at r2cloud
3. web ui
4. OS tasks

k7hak, 2021-01-19:
There was only a single dongle attached at a time, I swapped between them for troubleshooting reasons. I also don't believe its a problem with r2cloud but I feel that rp4 may be not compatible with rtl-sdrs at this time, I had a similar problem with the latest buster trying to get goestools to work as well. I totally dig r2cloud and was trying to gain some processing advantages from more ram and cpu cycles.

dernasherbrezon, 2021-01-23:
Got it. Yes, r2cloud depends on several natively-built applications. I can build and upload them for arm64.

dernasherbrezon, 2022-01-17:
I was able to successfully launch r2cloud on Pi4. No issues so far.

closed 2022-01-17 by dernasherbrezon