dernasherbrezon/r2cloud #109: Better quality picture

F21467, 2021-05-18:
Hello, can we get a better picture for NOAA and if so how? Thank you
[image]

Lukas0025, 2021-05-19:
it looks like you have a lot of noise in the picture. Better antenna or some filter can help. But with NOAA APT it is necessary to 
count that the images will not be the best, it is an analog transmission. Demonstration of standard quality on https://r2public.eu/observation?ob=319844 If you want good picture look on digital satellites like METEOR-M2 or HRPT of NOAA, but HRPT not supported by R2Cloud yet.

F21467, 2021-05-19:
Thank you for your answer

F21467, 2021-05-25:
With a new antenna it's OK

closed 2021-05-25 by F21467