dernasherbrezon/r2cloud #110: Unexpected closing

F21467, 2021-06-24:
Hello, I can open r2cloud after typing my login and my password, but as soon as I want to navigate to Observations or another tab, the software will close and allow me to log in and so on.
Do you have any idea why?

dernasherbrezon, 2021-06-25:
Several users reported the same issue. But there were not many changes recently to r2cloud-ui and I cannot reproduce it.

Most likely something happened with new browser's release.

Can you check the following:

1. r2cloud server logs. They are on raspberry pi: journalctl -u r2cloud -n 200
2. r2cloud UI logs. They are in browser: hit f12 and open dev console. Login and click on the “Observation” page. Check anything in the console output.
3. check r2cloud server response: 
   a. login into r2server
   b. open dev console and “Network” tab
   c. click on “Observation” page.
   d. check the response that was sent from server
4. Right after login, forcefully logout. I.e. login and then click on “logout” button. This should clear r2cloud token.

What browser are you using?

F21467, 2021-06-25:
Hello,

I use CHROME

I stopped the Raspberry and reinstalled everything after clearing all
browser history and cookies and it works again

Maybe my worries come from the fact that sometimes I don't use logout to
quit r2cloud but I close the page directly

Thank you for your help

cordially

R.FERNANDEZ

ea5wa, 2021-06-25:
Same here... Using Chrome in windows and Mozilla in Mageia Linux....

[image]

dernasherbrezon, 2021-06-26:
Ok, I found the issue:

1. Github created a pull request with the non-backward compatible library
2. This pull request was built
3. Travis-ci changed the master branch variable meaning. i.e. now it can include all pull request builds, so travis-ci published newly built (but not merged) .deb artefact globally.
4. This new .deb artefact r2cloud-ui had non-backward compatible library which was breaking authorisation.

I rolled the library back and re-released .deb artefact. Please:

```
sudo apt-get update
sudo apt-get install r2cloud-ui
```

closed 2021-06-26 by dernasherbrezon

F21467, 2021-06-26:
Thank You very much