dernasherbrezon/r2cloud #112: 502  r2cloud is not running

Eminlin, 2021-08-08:
`remove r2cloud` and `install r2cloud`, it doesn't work :(  

keyword: `Failed at step EXEC spawning /home/pi/r2cloud-jdk/bin/java: No such file or directory`

```
Aug 08 15:42:50 raspberrypi-2b systemd[2451]: r2cloud.service: Failed at step EXEC spawning /home/pi/r2cloud-jdk/bin/java: No such file or directory
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Main process exited, code=exited, status=203/EXEC
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Failed with result 'exit-code'.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Service RestartSec=100ms expired, scheduling restart.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Scheduled restart job, restart counter is at 3.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: Stopped R2Cloud Service.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: Started R2Cloud Service.
Aug 08 15:42:50 raspberrypi-2b systemd[2452]: r2cloud.service: Failed to execute command: No such file or directory
Aug 08 15:42:50 raspberrypi-2b systemd[2452]: r2cloud.service: Failed at step EXEC spawning /home/pi/r2cloud-jdk/bin/java: No such file or directory
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Main process exited, code=exited, status=203/EXEC
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Failed with result 'exit-code'.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Service RestartSec=100ms expired, scheduling restart.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Scheduled restart job, restart counter is at 4.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: Stopped R2Cloud Service.
Aug 08 15:42:50 raspberrypi-2b systemd[1]: Started R2Cloud Service.
Aug 08 15:42:50 raspberrypi-2b systemd[2453]: r2cloud.service: Failed to execute command: No such file or directory
Aug 08 15:42:50 raspberrypi-2b systemd[2453]: r2cloud.service: Failed at step EXEC spawning /home/pi/r2cloud-jdk/bin/java: No such file or directory
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Main process exited, code=exited, status=203/EXEC
Aug 08 15:42:50 raspberrypi-2b systemd[1]: r2cloud.service: Failed with result 'exit-code'.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: r2cloud.service: Service RestartSec=100ms expired, scheduling restart.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: r2cloud.service: Scheduled restart job, restart counter is at 5.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: Stopped R2Cloud Service.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: r2cloud.service: Start request repeated too quickly.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: r2cloud.service: Failed with result 'exit-code'.
Aug 08 15:42:51 raspberrypi-2b systemd[1]: Failed to start R2Cloud Service.
```

```
pi@raspberrypi-2b:~ $ ls -lh /usr/share/java/r2cloud/ | wc -l
26
```

dernasherbrezon, 2021-08-08:
Yes, looks like r2cloud-jdk was not fully installed. Can you explicitly install it using:

```
sudo apt-get install r2cloud-jdk
```

Eminlin, 2021-08-08:
> ```
> sudo apt-get install r2cloud-jdk
> ```

👀    
It seems to have been completely installed, but it still can't run

```shell
pi@raspberrypi-2b:~/r2cloudGithub $ sudo apt-get install r2cloud-jdk
Reading package lists... Done
Building dependency tree       
Reading state information... Done
r2cloud-jdk is already the newest version (11.0.10-1).
r2cloud-jdk set to manually installed.
The following packages were automatically installed and are no longer required:
  iso-codes libgd3 libncurses5 libnginx-mod-http-auth-pam libnginx-mod-http-dav-ext libnginx-mod-http-geoip libnginx-mod-http-image-filter
  libnginx-mod-http-subs-filter libnginx-mod-http-upstream-fair libnginx-mod-http-xslt-filter libnginx-mod-mail libnginx-mod-stream libpci3 libxpm4
  libxslt1.1 python-apt-common python3-apt python3-debconf
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 186 not upgraded.
```

dernasherbrezon, 2021-08-08:
```/home/pi/r2cloud-jdk/bin/java``` is a java binary. If you don't have that binary, then r2cloud-jdk was incorrectly installed. Can you remove it and install? I.e.

```
sudo apt-get remove r2cloud-jdk
sudo apt-get install r2cloud-jdk
```

Eminlin, 2021-08-08:
> `/home/pi/r2cloud-jdk/bin/java` is a java binary. If you don't have that binary, then r2cloud-jdk was incorrectly installed. Can you remove it and install? I.e.
> 
> ```
> sudo apt-get remove r2cloud-jdk
> sudo apt-get install r2cloud-jdk
> ```

I found that my java environment was lost. After redeploying the Java environment and trying the above command, the original project will also be removed, so the whole project needs to be reinstalled and can run now. I'm sorry for my stupid.  
  
Perhaps you can remind `java - version` in the installation document.  

Thank you again ☕

closed 2021-08-08 by Eminlin