dernasherbrezon/r2cloud #122: Observations overlaping with two SDR's

ArgoNavi, 2022-06-12:
My ground station consists of a rotator with two antennas (UHF+VHF) connected to separate SDR's, everything is working great except the station will occasionally schedule two overlapping observations on the separate receivers and then attempt to record both at the same time.

dernasherbrezon, 2022-06-13:
This is actually expected. Each rtl-sdr device is independent. r2cloud creates 2 separate schedules and load balance satellite observations between them. Most probably some observations can be scheduled in parallel.

Hmmm. Can you give more details about your hardware/software setup? What were your expectations?

ArgoNavi, 2022-06-13:
Just realized that r2cloud with one sdr connected doesn't end an observation early or start one later when the overlap is small, it simply chooses the higher priority satellite of the two and ignores the other.

This approach would also work fine for my two sdr setup.

dernasherbrezon, 2022-06-13:
I see. Essentially you want to re-use rotator for 2 different feeds.

I fear this setup is not supported yet. In this case I have to introduce some kind of logical pair of physical devices. Or create exclusive sdr devices. In that case satellites from 2 physical devices will be scheduled without overlap.

I guess the rotator with 2 different feeds is quite common. Right?

ArgoNavi, 2022-06-13:
Yes that's exactly what I want, hopefully its possible to implement and not too difficult for you.

Most people with a rotator have 2 different feeds if not more. However they use a single sdr with a complicated setup involving coax switches and scrips to select the proper antenna before an observation.

https://wiki.satnogs.org/Antenna_switching

I couldn't figure out how to do this which is why I switched from using satnogs to r2cloud since it supports using multiple sdrs.

dernasherbrezon, 2022-06-22:
I implemented this feature using a shared schedule. I.e. if any 2 or more devices have the same rotator configuration (host + port), then they will share the same schedule.
https://github.com/dernasherbrezon/r2cloud/blob/master/src/test/java/ru/r2cloud/it/SharedRotatorTest.java

closed 2022-06-22 by dernasherbrezon