dernasherbrezon/r2cloud #123: Dissapearing Observations

ArgoNavi, 2022-07-20:
Observations scheduled to start immediately after another is finished disappear from the list at the time they should begin, the rotator tracks the satellite but no recording are done and no data is created.

If I restart r2cloud with "sudo systemctl restart r2cloud" the observation starts and proceeds as normal except shorter in length.

ArgoNavi, 2022-07-20:
Found the relevent area of the log, ELFIN-B stops but GOMX-1 failed to start untill I rest r2cloud.

>
Jul 20 01:25:12 r2cloud java[1349]: [sdr-0] scheduled next pass for 43768-0. start: Fri Jul 22 03:46:43 UTC 2022 end: Fri Jul 22 03:53:17 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: satellite ELFIN-B(43616) enabled [ru.r2cloud.device.DeviceManager enableSatellite]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Wed Jul 20 05:31:43 UTC 2022 end: Wed Jul 20 05:36:02 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Wed Jul 20 17:12:33 UTC 2022 end: Wed Jul 20 17:19:44 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Fri Jul 22 04:49:37 UTC 2022 end: Fri Jul 22 04:56:17 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:30:59 r2cloud java[1349]: [1658293993190-43880-0] decoded [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Jul 20 01:30:59 r2cloud java[1349]: [1658293993190-43880-0] uploading observation [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Jul 20 01:31:02 r2cloud java[1349]: [1658293993190-43880-0] generating spectogram [ru.r2cloud.SpectogramService create]
Jul 20 01:31:43 r2cloud java[1349]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437496397 -d 0 -s 240000 -g 15.0 -p 0 -o /tmp/43616-1658295103271-43616-0.raw.gz [ru.r2cloud.util.ProcessFacto
Jul 20 01:31:43 r2cloud java[1349]: [1658295103271-43616-0] moving rotator to Position [azimuth=354.7536786597215, elevation=4.952321139282463] [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:31:43 r2cloud java[1349]: Found 1 device(s):
Jul 20 01:31:43 r2cloud java[1349]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:31:43 r2cloud java[1349]: Using device 0: Generic RTL2832U OEM
Jul 20 01:31:43 r2cloud java[1349]: Found Rafael Micro R820T tuner
Jul 20 01:31:43 r2cloud java[1349]: [R82XX] PLL not locked!
Jul 20 01:31:43 r2cloud java[1349]: Sampling at 240000 S/s.
Jul 20 01:31:43 r2cloud java[1349]: Tuned to 437496397 Hz.
Jul 20 01:31:44 r2cloud java[1349]: Tuner gain set to 14.40 dB.
Jul 20 01:31:44 r2cloud java[1349]: Reading samples in async mode...
Jul 20 01:34:21 r2cloud java[1349]: [1658293993190-43880-0] spectogram created [ru.r2cloud.SpectogramService create]
Jul 20 01:34:21 r2cloud java[1349]: [1658293993190-43880-0] observation uploaded [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Jul 20 01:36:02 r2cloud java[1349]: [1658295362726-39430-0] moving rotator to Position [azimuth=354.49521146945466, elevation=5.00010026617825] [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:36:02 r2cloud java[1349]: stopping: rtl_sdr for 1658295103271-43616-0 [ru.r2cloud.util.Util shutdown]
Jul 20 01:36:02 r2cloud java[1349]: Signal caught, exiting!
Jul 20 01:36:02 r2cloud java[1349]: [1658295103271-43616-0] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jul 20 01:36:02 r2cloud java[1349]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437268929 -d 0 -s 240000 -g 15.0 -p 0 -o /tmp/39430-1658295362726-39430-0.raw.gz [ru.r2cloud.util.ProcessFacto
Jul 20 01:36:02 r2cloud java[1349]: [1658295103271-43616-0] decoding [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Jul 20 01:36:02 r2cloud java[1349]: Found 1 device(s):
Jul 20 01:36:02 r2cloud java[1349]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:36:02 r2cloud java[1349]: Using device 0: Generic RTL2832U OEM
Jul 20 01:36:02 r2cloud java[1349]: usb_claim_interface error -6
Jul 20 01:36:02 r2cloud java[1349]: Failed to open rtlsdr device #0.
Jul 20 01:36:02 r2cloud java[1349]: [1658295362726-39430-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jul 20 01:36:03 r2cloud java[1349]: [1658295103271-43616-0] observation time passed. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:36:03 r2cloud java[1349]: User cancel, exiting...
Jul 20 01:38:07 r2cloud systemd[1]: Stopping R2Cloud Service...
Jul 20 01:38:07 r2cloud java[1349]: stopping [ru.r2cloud.R2Cloud$1 run]
Jul 20 01:38:07 r2cloud java[1349]: [rotator-127.0.0.1:4533] stopped [ru.r2cloud.satellite.RotatorService stop]
Jul 20 01:38:07 r2cloud java[1349]: [sdr-0] device stopped [ru.r2cloud.device.Device stop]
Jul 20 01:38:12 r2cloud java[1349]: executor did not terminate in the specified time: decoder [ru.r2cloud.util.Util shutdown]
Jul 20 01:38:12 r2cloud java[1349]: =========== stopped ============= [ru.r2cloud.R2Cloud$1 run]
Jul 20 01:38:12 r2cloud systemd[1]: r2cloud.service: Succeeded.
Jul 20 01:38:12 r2cloud systemd[1]: Stopped R2Cloud Service.
Jul 20 01:38:12 r2cloud systemd[1]: Started R2Cloud Service.

Jul 20 01:39:09 r2cloud java[1704]: total memory: 25833472 [ru.r2cloud.metrics.Metrics start]
Jul 20 01:39:09 r2cloud java[1704]: CPU count: 4 [ru.r2cloud.metrics.Metrics start]
Jul 20 01:39:09 r2cloud java[1704]: Found 1 device(s):
Jul 20 01:39:09 r2cloud java[1704]: SIGAR library was loaded [ru.r2cloud.metrics.Metrics start]
Jul 20 01:39:09 r2cloud java[1704]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:39:09 r2cloud java[1704]: Using device 0: Generic RTL2832U OEM
Jul 20 01:39:09 r2cloud java[1704]: metrics started [ru.r2cloud.metrics.Metrics start]
[... cut ...]
Jul 20 02:02:37 r2cloud java[1704]: Found 1 device(s):
Jul 20 02:02:37 r2cloud java[1704]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 02:02:37 r2cloud java[1704]: Using device 0: Generic RTL2832U OEM
Jul 20 02:02:37 r2cloud java[1704]: Found Rafael Micro R820T tuner
Jul 20 02:02:37 r2cloud java[1704]: [R82XX] PLL not locked!
Jul 20 02:02:38 r2cloud java[1704]: Sampling at 240000 S/s.
Jul 20 02:02:38 r2cloud java[1704]: Tuned to 145849066 Hz.
Jul 20 02:02:38 r2cloud java[1704]: Tuner gain set to 14.40 dB.

ArgoNavi, 2022-07-20:
```shell

-- Logs begin at Tue 2022-06-28 00:54:21 EDT, end at Wed 2022-07-20 13:23:29 EDT. --
Jul 20 01:13:13 r2cloud java[1349]: Found 1 device(s):
Jul 20 01:13:13 r2cloud java[1349]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:13:13 r2cloud java[1349]: Using device 0: Generic RTL2832U OEM
Jul 20 01:13:13 r2cloud java[1349]: Found Rafael Micro R820T tuner
Jul 20 01:13:14 r2cloud java[1349]: [R82XX] PLL not locked!
Jul 20 01:13:14 r2cloud java[1349]: Sampling at 240000 S/s.
Jul 20 01:13:14 r2cloud java[1349]: Tuned to 435620016 Hz.
Jul 20 01:13:14 r2cloud java[1349]: Tuner gain set to 14.40 dB.
Jul 20 01:13:14 r2cloud java[1349]: Reading samples in async mode...
Jul 20 01:17:27 r2cloud java[1349]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Jul 20 01:17:34 r2cloud java[1349]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Jul 20 01:23:00 r2cloud java[1349]: stopping: rtl_sdr for 1658293993190-43880-0 [ru.r2cloud.util.Util shutdown]
Jul 20 01:23:01 r2cloud java[1349]: Signal caught, exiting!
Jul 20 01:23:01 r2cloud java[1349]: [1658293993190-43880-0] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jul 20 01:23:01 r2cloud java[1349]: [1658293993190-43880-0] observation time passed. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:23:01 r2cloud java[1349]: deleting old observation: ./data/satellites/44352/data/1657279766840-44352-0 last update time: Fri Jul 08 11:36:40 UTC 2022 [ru.r2cloud.satellite.TimeSizeRetention indexAndCle
Jul 20 01:23:01 r2cloud java[1349]: [1658293993190-43880-0] decoding [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Jul 20 01:23:01 r2cloud java[1349]: User cancel, exiting...
Jul 20 01:25:11 r2cloud java[1349]: satellite AISTECHSAT 2(43768) enabled [ru.r2cloud.device.DeviceManager enableSatellite]
Jul 20 01:25:12 r2cloud java[1349]: [sdr-0] scheduled next pass for 43768-0. start: Fri Jul 22 03:46:43 UTC 2022 end: Fri Jul 22 03:53:17 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: satellite ELFIN-B(43616) enabled [ru.r2cloud.device.DeviceManager enableSatellite]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Wed Jul 20 05:31:43 UTC 2022 end: Wed Jul 20 05:36:02 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Wed Jul 20 17:12:33 UTC 2022 end: Wed Jul 20 17:19:44 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:25:17 r2cloud java[1349]: [sdr-0] scheduled next pass for 43616-0. start: Fri Jul 22 04:49:37 UTC 2022 end: Fri Jul 22 04:56:17 UTC 2022 [ru.r2cloud.device.Device schedule]
Jul 20 01:30:59 r2cloud java[1349]: [1658293993190-43880-0] decoded [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Jul 20 01:30:59 r2cloud java[1349]: [1658293993190-43880-0] uploading observation [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Jul 20 01:31:02 r2cloud java[1349]: [1658293993190-43880-0] generating spectogram [ru.r2cloud.SpectogramService create]
Jul 20 01:31:43 r2cloud java[1349]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437496397 -d 0 -s 240000 -g 15.0 -p 0 -o /tmp/43616-1658295103271-43616-0.raw.gz [ru.r2cloud.util.ProcessFacto
Jul 20 01:31:43 r2cloud java[1349]: [1658295103271-43616-0] moving rotator to Position [azimuth=354.7536786597215, elevation=4.952321139282463] [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:31:43 r2cloud java[1349]: Found 1 device(s):
Jul 20 01:31:43 r2cloud java[1349]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:31:43 r2cloud java[1349]: Using device 0: Generic RTL2832U OEM
Jul 20 01:31:43 r2cloud java[1349]: Found Rafael Micro R820T tuner
Jul 20 01:31:43 r2cloud java[1349]: [R82XX] PLL not locked!
Jul 20 01:31:43 r2cloud java[1349]: Sampling at 240000 S/s.
Jul 20 01:31:43 r2cloud java[1349]: Tuned to 437496397 Hz.
Jul 20 01:31:44 r2cloud java[1349]: Tuner gain set to 14.40 dB.
Jul 20 01:31:44 r2cloud java[1349]: Reading samples in async mode...
Jul 20 01:34:21 r2cloud java[1349]: [1658293993190-43880-0] spectogram created [ru.r2cloud.SpectogramService create]
Jul 20 01:34:21 r2cloud java[1349]: [1658293993190-43880-0] observation uploaded [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Jul 20 01:36:02 r2cloud java[1349]: [1658295362726-39430-0] moving rotator to Position [azimuth=354.49521146945466, elevation=5.00010026617825] [ru.r2cloud.satellite.RotatorService$1 run]
Jul 20 01:36:02 r2cloud java[1349]: stopping: rtl_sdr for 1658295103271-43616-0 [ru.r2cloud.util.Util shutdown]
Jul 20 01:36:02 r2cloud java[1349]: Signal caught, exiting!
Jul 20 01:36:02 r2cloud java[1349]: [1658295103271-43616-0] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jul 20 01:36:02 r2cloud java[1349]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437268929 -d 0 -s 240000 -g 15.0 -p 0 -o /tmp/39430-1658295362726-39430-0.raw.gz [ru.r2cloud.util.ProcessFacto
Jul 20 01:36:02 r2cloud java[1349]: [1658295103271-43616-0] decoding [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Jul 20 01:36:02 r2cloud java[1349]: Found 1 device(s):
Jul 20 01:36:02 r2cloud java[1349]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 01:36:02 r2cloud java[1349]: Using device 0: Generic RTL2832U OEM
Jul 20 01:36:02 r2cloud java[1349]: usb_claim_interface error -6
[... cut ...]
Jul 20 02:02:37 r2cloud java[1704]: Found 1 device(s):
Jul 20 02:02:37 r2cloud java[1704]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jul 20 02:02:37 r2cloud java[1704]: Using device 0: Generic RTL2832U OEM
Jul 20 02:02:37 r2cloud java[1704]: Found Rafael Micro R820T tuner
Jul 20 02:02:37 r2cloud java[1704]: [R82XX] PLL not locked!
Jul 20 02:02:38 r2cloud java[1704]: Sampling at 240000 S/s.
Jul 20 02:02:38 r2cloud java[1704]: Tuned to 145849066 Hz.
Jul 20 02:02:38 r2cloud java[1704]: Tuner gain set to 14.40 dB.
lines 233-303

ArgoNavi, 2022-07-20:
Seems like its only happening when a new observation starts within seconds of another finishing, maybe introduce a delay between observations?

ArgoNavi, 2022-07-24:
Last commit fixed the problem, Thanks!

ArgoNavi, 2023-06-28:
Started happening again, I think "scheduler.data.retention.maxSizeBytes" no longer works.

```shell

Jun 28 08:08:16 r2cloud java[3005]: Found 1 device(s):
Jun 28 08:08:16 r2cloud java[3005]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 28 08:08:16 r2cloud java[3005]: Using device 0: Generic RTL2832U OEM
Jun 28 08:08:16 r2cloud java[3005]: Found Rafael Micro R820T tuner
Jun 28 08:08:16 r2cloud java[3005]: [R82XX] PLL not locked!
Jun 28 08:08:16 r2cloud java[3005]: Sampling at 240000 S/s.
Jun 28 08:08:16 r2cloud java[3005]: Tuned to 435100280 Hz.
Jun 28 08:08:16 r2cloud java[3005]: Tuner gain set to 44.50 dB.
Jun 28 08:08:16 r2cloud java[3005]: Reading samples in async mode...
Jun 28 08:09:39 r2cloud java[3005]: loading tle from: http://celestrak.com/NORAD/elements/satnogs.txt [ru.r2cloud.tle.CelestrakClient downloadTle]
Jun 28 08:09:40 r2cloud java[3005]: received tle for 228 satellites [ru.r2cloud.tle.CelestrakClient downloadTle]
Jun 28 08:09:40 r2cloud java[3005]: loading tle from: http://celestrak.com/NORAD/elements/active.txt [ru.r2cloud.tle.CelestrakClient downloadTle]
Jun 28 08:09:40 r2cloud java[3005]: received tle for 8012 satellites [ru.r2cloud.tle.CelestrakClient downloadTle]
Jun 28 08:09:41 r2cloud java[3005]: check for not processed observations [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Jun 28 08:15:27 r2cloud java[3005]: gzip: stdout: No space left on device
Jun 28 08:15:29 r2cloud java[3005]: Signal caught, exiting!
Jun 28 08:15:29 r2cloud java[3005]: Short write, samples lost, exiting!
Jun 28 08:15:29 r2cloud java[3005]: User cancel, exiting...
Jun 28 08:15:29 r2cloud java[3005]: [1687954096242-51658-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jun 28 08:15:30 r2cloud java[3005]: [1687954096242-51658-0] observation stopped. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 08:25:47 r2cloud java[3005]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437519771 -d 0 -s 240000 -g 45.0 -p 0 -o /tmp/51657-1687955147174-51657-0.raw.gz [ru.r2cloud.util.Proce
Jun 28 08:25:47 r2cloud java[3005]: [1687955147174-51657-0] moving rotator to Position [azimuth=355.8279068474617, elevation=4.996769910471345] [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 08:25:47 r2cloud java[3005]: Found 1 device(s):
Jun 28 08:25:47 r2cloud java[3005]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 28 08:25:47 r2cloud java[3005]: Using device 0: Generic RTL2832U OEM
Jun 28 08:25:47 r2cloud java[3005]: Found Rafael Micro R820T tuner
Jun 28 08:25:47 r2cloud java[3005]: [R82XX] PLL not locked!
Jun 28 08:25:47 r2cloud java[3005]: Sampling at 240000 S/s.
Jun 28 08:25:47 r2cloud java[3005]: Tuned to 437519771 Hz.
Jun 28 08:25:47 r2cloud java[3005]: Tuner gain set to 44.50 dB.
Jun 28 08:25:47 r2cloud java[3005]: Reading samples in async mode...
Jun 28 08:33:07 r2cloud java[3005]: gzip: stdout: No space left on device
Jun 28 08:33:08 r2cloud java[3005]: Signal caught, exiting!
Jun 28 08:33:08 r2cloud java[3005]: Short write, samples lost, exiting!
Jun 28 08:33:09 r2cloud java[3005]: User cancel, exiting...
Jun 28 08:33:09 r2cloud java[3005]: [1687955147174-51657-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jun 28 08:33:10 r2cloud java[3005]: [1687955147174-51657-0] observation stopped. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 08:38:08 r2cloud java[3005]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437465206 -d 0 -s 240000 -g 45.0 -p 0 -o /tmp/40014-1687955888739-40014-0.raw.gz [ru.r2cloud.util.Proce
Jun 28 08:38:08 r2cloud java[3005]: [1687955888739-40014-0] moving rotator to Position [azimuth=9.089296816884648, elevation=4.989666479277279] [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 08:38:08 r2cloud java[3005]: Found 1 device(s):
Jun 28 08:38:08 r2cloud java[3005]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Jun 28 08:38:08 r2cloud java[3005]: Using device 0: Generic RTL2832U OEM
Jun 28 08:38:09 r2cloud java[3005]: Found Rafael Micro R820T tuner
Jun 28 08:38:09 r2cloud java[3005]: [R82XX] PLL not locked!
Jun 28 08:38:09 r2cloud java[3005]: Sampling at 240000 S/s.
Jun 28 08:38:09 r2cloud java[3005]: Tuned to 437465206 Hz.
Jun 28 08:38:09 r2cloud java[3005]: Tuner gain set to 44.50 dB.
Jun 28 08:38:09 r2cloud java[3005]: Reading samples in async mode...
Jun 28 08:45:41 r2cloud java[3005]: gzip: stdout: No space left on device
Jun 28 08:45:42 r2cloud java[3005]: Signal caught, exiting!
Jun 28 08:45:42 r2cloud java[3005]: Short write, samples lost, exiting!
Jun 28 08:45:43 r2cloud java[3005]: User cancel, exiting...
Jun 28 08:45:43 r2cloud java[3005]: [1687955888739-40014-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Jun 28 08:45:43 r2cloud java[3005]: [1687955888739-40014-0] observation stopped. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 09:06:32 r2cloud java[3005]: [1687957592515-25338-0] moving rotator to Position [azimuth=13.698097687128218, elevation=5.000093320722774] [ru.r2cloud.satellite.RotatorService$1 run]
Jun 28 09:06:32 r2cloud java[3005]: started with arguments: sox -t raw -r 60000 -es -b 16 - /tmp/25338-1687957592515-25338-0.wav rate 11025 [ru.r2cloud.util.ProcessFactory create]
Jun 28 09:06:32 r2cloud java[3005]: started with arguments: rtl_fm -f 137620000 -d 0 -s 60000 -g 45.0 -p 0 -E deemp -F 9 - [ru.r2cloud.util.ProcessFactory create]
Jun 28 09:06:32 r2cloud java[3005]: Found 1 device(s):
lines 493-551

dernasherbrezon, 2023-06-28:
It seems there is no disk space left on the device:

```
Jun 28 08:15:27 r2cloud java[3005]: gzip: stdout: No space left on device
```

r2cloud can manage only it's own data. Thus ```keep last 25558Mb of observations``` actually means it start deleting observation only when observations exceed 25558Mb. If someone else is occupying the disk, then r2cloud won't delete observations in order to free up the space. This makes size of observations folder stable and guaranteed. Can you check how many bytes observations are actually take?

```
du -h ~/r2cloud/data
```