dernasherbrezon/r2cloud #128: Missing TLE

ArgoNavi, 2022-09-07:
I noticed that most of the new satellites never get scheduled and after checking it seems they lack TLE

```shell
Sep 07 02:11:42 r2cloud java[15245]: [sdr-0] device started [ru.r2cloud.device.Device start]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: ABXM-4898-9222-6959-5721 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: AGPD-5250-3566-0713-3550 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: BDML-1083-7096-5526-2883 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: CCNX-9563-2937-7997-4713 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: CIOM-1903-7117-1067-8603 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: EAHI-9090-2555-3685-8378 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: FWFR-1086-4147-8825-4245 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: HNHF-6321-7258-0988-3559 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: ISLM-2525-1913-5885-2577 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: MADO-9521-7061-1658-0034 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: MKYL-1174-0802-7826-9086 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: OSBB-5160-7827-1540-1828 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: QZQH-5328-1484-7427-0152 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: QZYF-0302-7653-1068-1013 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: SXFL-2149-3788-0081-7121 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: no tle for: TQIP-5428-9682-2220-6430 [ru.r2cloud.satellite.ObservationFactory createSchedule]
Sep 07 02:12:00 r2cloud java[15245]: HIGH: satellites 16 observations 0 [ru.r2cloud.satellite.Schedule scheduleSatellites]
Sep 07 02:12:00 r2cloud java[15245]: NORMAL: satellites 117 observations 189 [ru.r2cloud.satellite.Schedule scheduleSatellites]

ArgoNavi, 2022-09-08:
Now my TLE's say they are one week out of date and not updating despite being fine a couple of hours ago.
[image]

dernasherbrezon, 2022-09-08:
These are satnogs satellites. Satnogs api provides TLE in a separate API call. And this separate call might failed. Can you please check there were no errors before that?

ArgoNavi, 2022-09-08:
That’s weird as I don’t have satnogs enabled, should I not see these satellites in my list?

dernasherbrezon, 2022-09-08:
Yes you shouldn't see them. These id format is used in Satnogs. I had to configure it in leosatdata.com also to match with satnogs. "new launches" had these ids for couple of days only. There were also 2 APIs for "new launches" at leosatdata.com during the migration. It seems you might hit the migration issue.

Do you see this issue right now? These satellites have permanent NORAD ids now.

ArgoNavi, 2022-09-08:
Unfortunately I had to re-flash and start over to fix my problem.

ArgoNavi, 2022-09-10:
Still having the problem of my TLE's not updating. 

I think it might be a problem with leosatdata.com as after adding my API key XW-2A/B/C are added to my list but no tle for them is added, "new launches" is disabled so I'm not sure why its adding satellites.

ArgoNavi, 2022-09-10:
``` shell
Sep 10 07:28:58 r2cloud java[533]: =========== stopped ============= [ru.r2cloud.R2Cloud$1 run]
Sep 10 07:28:58 r2cloud systemd[1]: r2cloud.service: Succeeded.
Sep 10 07:28:58 r2cloud systemd[1]: Stopped R2Cloud Service.
Sep 10 07:28:58 r2cloud systemd[1]: Started R2Cloud Service.
Sep 10 07:29:01 r2cloud java[703]: retention: keep last 25600Mb of observations [ru.r2cloud.satellite.ObservationDao <init>]
Sep 10 07:29:07 r2cloud java[703]: HIGH: 0 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 10 07:29:07 r2cloud java[703]: NORMAL: 141 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 10 07:29:07 r2cloud java[703]:   CONFIG: 141 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 10 07:29:07 r2cloud java[703]: ddns is disabled [ru.r2cloud.ddns.DDNSClient start]
Sep 10 07:29:07 r2cloud java[703]: check for not processed observations [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Sep 10 07:29:07 r2cloud java[703]: [rotator-127.0.0.1:4533] starting rotator on: 127.0.0.1:4533 [ru.r2cloud.satellite.RotatorService start]
Sep 10 07:29:07 r2cloud java[703]: [rotator-127.0.0.1:4533] initialized for model: EasycommIII [ru.r2cloud.satellite.RotatorService ensureClientConnected]
Sep 10 07:29:07 r2cloud java[703]: [sdr-0] device started [ru.r2cloud.device.Device start]
Sep 10 07:29:21 r2cloud java[703]: HIGH: satellites 0 observations 0 [ru.r2cloud.satellite.Schedule scheduleSatellites]
Sep 10 07:29:25 r2cloud java[703]: NORMAL: satellites 67 observations 135 [ru.r2cloud.satellite.Schedule scheduleSatellites]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 39090-0. start: Sat Sep 10 11:42:44 UTC 2022 end: Sat Sep 10 11:54:58 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 40043-0. start: Sat Sep 10 11:56:11 UTC 2022 end: Sat Sep 10 12:05:40 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 51658-0. start: Sat Sep 10 12:08:55 UTC 2022 end: Sat Sep 10 12:17:12 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 51657-0. start: Sat Sep 10 12:17:12 UTC 2022 end: Sat Sep 10 12:26:01 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 30776-0. start: Sat Sep 10 12:32:46 UTC 2022 end: Sat Sep 10 12:40:30 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 25338-0. start: Sat Sep 10 13:07:59 UTC 2022 end: Sat Sep 10 13:20:39 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 40379-0. start: Sat Sep 10 13:20:39 UTC 2022 end: Sat Sep 10 13:25:31 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 42775-0. start: Sat Sep 10 13:35:32 UTC 2022 end: Sat Sep 10 13:44:34 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 33591-0. start: Sat Sep 10 14:00:38 UTC 2022 end: Sat Sep 10 14:13:41 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 30776-0. start: Sat Sep 10 14:13:41 UTC 2022 end: Sat Sep 10 14:17:55 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 52950-0. start: Sat Sep 10 14:18:45 UTC 2022 end: Sat Sep 10 14:28:10 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43908-0. start: Sat Sep 10 14:34:50 UTC 2022 end: Sat Sep 10 14:42:50 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43616-0. start: Sat Sep 10 14:48:16 UTC 2022 end: Sat Sep 10 14:52:30 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 39469-0. start: Sat Sep 10 14:52:30 UTC 2022 end: Sat Sep 10 15:04:02 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 47954-0. start: Sat Sep 10 15:14:03 UTC 2022 end: Sat Sep 10 15:22:33 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 28654-0. start: Sat Sep 10 15:27:47 UTC 2022 end: Sat Sep 10 15:40:20 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 53379-0. start: Sat Sep 10 15:40:35 UTC 2022 end: Sat Sep 10 15:48:53 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43770-0. start: Sat Sep 10 15:50:36 UTC 2022 end: Sat Sep 10 16:00:35 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43132-0. start: Sat Sep 10 16:05:48 UTC 2022 end: Sat Sep 10 16:14:21 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43793-0. start: Sat Sep 10 16:14:21 UTC 2022 end: Sat Sep 10 16:22:37 UTC 2022 [ru.r2cloud.device.Device schedule]
Sep 10 07:29:25 r2cloud java[703]: [sdr-0] scheduled next pass for 43880-0. start: Sat Sep 10 16:23:24 UTC 2022 end: Sat Sep 10 16:33:19 UTC 2022 [ru.r2cloud.device.Device schedule]
[... cut ...]
Sep 10 07:42:44 r2cloud java[703]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Sep 10 07:42:44 r2cloud java[703]: Using device 0: Generic RTL2832U OEM
Sep 10 07:42:44 r2cloud java[703]: Found Rafael Micro R820T tuner
Sep 10 07:42:45 r2cloud java[703]: [R82XX] PLL not locked!
Sep 10 07:42:45 r2cloud java[703]: Sampling at 240000 S/s.
Sep 10 07:42:45 r2cloud java[703]: Tuned to 437587812 Hz.
Sep 10 07:42:45 r2cloud java[703]: Tuner gain set to 28.00 dB.
Sep 10 07:42:45 r2cloud java[703]: Reading samples in async mode...

dernasherbrezon, 2022-09-10:
Found it. Here is what going on:

- r2cloud starts for the first time
- load hardcoded list of satellites
- load TLE for ~6k but stores only what r2cloud need. This is to avoid disk pollution with decayed TLEs
- you configure leosatdata integration and this will provide additional set of satellites
- on next startup r2cloud loads both hardcoded and leosatdata satellites
- but the local TLE cache file was not updated yet, thus no additional TLE for leosatdata satellites
- as a result: error in the log

There are several ways to fix this:

- Do not fix this. r2cloud will wait next ```housekeeping.tle.periodMillis``` and reload TLE automatically. Currently 2 days
- Reload from celestrak if at least one satellite doesn't have TLE. That was implemented previously. However if leosatdata or configs contain decayed satellite, then r2cloud will load the TLE from celestrak every time.

ArgoNavi, 2022-09-10:
Ok, I assumed TLE's where updated daily not every 2 days so thats what caused my confusion.

also what dose "satellites.tle.lastupdateAtMillis=1662806658060" in ~/.r2cloud mean?

dernasherbrezon, 2022-09-10:
This is when the TLE last updated. Previously each satellite had own "tle.txt" file. So this parameter was used to keep track of last update time.

closed 2022-09-10 by ArgoNavi