dernasherbrezon/r2cloud #129: Scheduler broken by update

ArgoNavi, 2022-09-15:
Scheduler stopped working and no observations are being done.

``` shell

Sep 15 17:52:29 r2cloud java[527]: ================================= [ru.r2cloud.R2Cloud start]
Sep 15 17:52:29 r2cloud java[527]: =========== started ============= [ru.r2cloud.R2Cloud start]
Sep 15 17:52:29 r2cloud java[527]: ================================= [ru.r2cloud.R2Cloud start]
Sep 15 17:53:12 r2cloud java[527]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 15 17:53:14 r2cloud java[527]: unable to handle request
Sep 15 17:53:14 r2cloud java[527]: java.lang.NullPointerException
Sep 15 17:53:14 r2cloud java[527]:         at ru.r2cloud.web.api.schedule.ScheduleList.doGet(ScheduleList.java:33)
Sep 15 17:53:14 r2cloud java[527]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 15 17:53:14 r2cloud java[527]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 15 17:53:14 r2cloud java[527]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 15 17:53:14 r2cloud java[527]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 15 17:53:14 r2cloud java[527]:  [ru.r2cloud.web.WebServer serve]
Sep 15 17:53:24 r2cloud java[527]: unable to handle request
Sep 15 17:53:24 r2cloud java[527]: java.lang.NullPointerException
Sep 15 17:53:24 r2cloud java[527]:         at ru.r2cloud.web.api.schedule.ScheduleList.doGet(ScheduleList.java:33)
Sep 15 17:53:24 r2cloud java[527]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 15 17:53:24 r2cloud java[527]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 15 17:53:24 r2cloud java[527]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 15 17:53:24 r2cloud java[527]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 15 17:53:24 r2cloud java[527]:  [ru.r2cloud.web.WebServer serve]

dernasherbrezon, 2022-09-16:
It looks like cached satellites from leosatdata.com are incorrect. Can you please:

```
rm /home/pi/r2cloud/data/leosatdata.*
```

And restart? In the meantime I will add enough safe checks to avoid this in the future.

closed 2022-09-16 by ArgoNavi

dernasherbrezon, 2022-09-19:
Fixed in bde457f7abc413363b1cc72410b4d3e42ee4a15b