dernasherbrezon/r2cloud #130: Meteor-M2

plato8888, 2022-09-24:
Please correct the frequency for the Meteor-M2 satellite.  The satellite uses 137.1Mhz for LRPT, not 137.9Mhz.  I need to manually correct the leosatdata.json file every few days to get the r2could service to listen in on the correct frequency to obtain the weather images. 

Thank you.

dernasherbrezon, 2022-09-24:
Fixed in [855dba0](https://github.com/dernasherbrezon/r2cloud/commit/855dba0bda5c0166524c1bf56bcc620cced3068e)

closed 2022-09-24 by dernasherbrezon

plato8888, 2022-09-24:
> Fixed in [855dba0](https://github.com/dernasherbrezon/r2cloud/commit/855dba0bda5c0166524c1bf56bcc620cced3068e)

Thank you!  I'll provide feedback in the next few days to confirm if it corrected the frequency used for future observations.

Take care.