dernasherbrezon/r2cloud #131: METEOR-M 2 observations not loading

ArgoNavi, 2022-09-25:
When I try to view my METEOR-M 2 observations I get "502 r2cloud is not running", checking the logs show several errors.

``` shell
Sep 25 10:35:16 r2cloud java[7379]: [1664111156309-39090-0] spectogram created [ru.r2cloud.SpectogramService create]
Sep 25 10:35:16 r2cloud java[7379]: [1664111156309-39090-0] observation uploaded [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Sep 25 10:35:16 r2cloud java[7379]: [1664113705135-42775-0] decoding [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Sep 25 10:37:12 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#5)
Sep 25 10:37:12 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.jradio.BeaconInputStream.readProtocolv1(BeaconInputStream.java:44)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.jradio.BeaconInputStream.hasNext(BeaconInputStream.java:27)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.api.observation.ObservationLoad.convertBeacons(ObservationLoad.java:93)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.api.observation.ObservationLoad.doGet(ObservationLoad.java:68)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 25 10:37:12 r2cloud java[7379]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 25 10:37:12 r2cloud java[7379]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 25 10:37:12 r2cloud java[7379]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 25 10:37:12 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 10:37:12 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#6)
Sep 25 10:37:12 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.jradio.BeaconInputStream.readProtocolv1(BeaconInputStream.java:44)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.jradio.BeaconInputStream.hasNext(BeaconInputStream.java:27)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.api.observation.ObservationLoad.convertBeacons(ObservationLoad.java:93)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.api.observation.ObservationLoad.doGet(ObservationLoad.java:68)
Sep 25 10:37:12 r2cloud java[7379]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 25 10:37:12 r2cloud java[7379]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 25 10:37:12 r2cloud java[7379]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 25 10:37:12 r2cloud java[7379]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 25 10:37:12 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 10:37:14 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#7)
Sep 25 10:37:14 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:14 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 10:37:31 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#10)
Sep 25 10:37:31 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:31 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 10:37:35 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#11)
Sep 25 10:37:35 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:35 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 10:37:35 r2cloud java[7379]: UncaughtException at: NanoHttpd Request Processor (#12)
Sep 25 10:37:35 r2cloud java[7379]: java.lang.OutOfMemoryError: Java heap space
Sep 25 10:37:35 r2cloud java[7379]:  [ru.r2cloud.R2Cloud$2 uncaughtException]

dernasherbrezon, 2022-09-25:
Fixed in [6c34eec](https://github.com/dernasherbrezon/r2cloud/commit/6c34eec4c9706972fa4af5368cfcdcdf59dac83b). LRPT has too much frames to load into memory. Thus failing with OOM.

closed 2022-09-25 by dernasherbrezon

ArgoNavi, 2022-09-25:
Still a problem even after the fix.

``` shell
Sep 25 13:24:47 r2cloud java[1064]: ================================= [ru.r2cloud.R2Cloud start]
Sep 25 13:24:47 r2cloud java[1064]: =========== started ============= [ru.r2cloud.R2Cloud start]
Sep 25 13:24:47 r2cloud java[1064]: ================================= [ru.r2cloud.R2Cloud start]
Sep 25 13:25:14 r2cloud java[1064]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 25 13:25:57 r2cloud java[1064]: UncaughtException at: NanoHttpd Request Processor (#5)
Sep 25 13:25:57 r2cloud java[1064]: java.lang.OutOfMemoryError: Java heap space
Sep 25 13:25:57 r2cloud java[1064]:         at ru.r2cloud.jradio.BeaconInputStream.readProtocolv1(BeaconInputStream.java:44)
Sep 25 13:25:57 r2cloud java[1064]:         at ru.r2cloud.jradio.BeaconInputStream.hasNext(BeaconInputStream.java:27)
Sep 25 13:25:57 r2cloud java[1064]:         at ru.r2cloud.web.api.observation.ObservationLoad.convertBeacons(ObservationLoad.java:93)
Sep 25 13:25:57 r2cloud java[1064]:         at ru.r2cloud.web.api.observation.ObservationLoad.doGet(ObservationLoad.java:68)
Sep 25 13:25:57 r2cloud java[1064]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 25 13:25:57 r2cloud java[1064]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 25 13:25:57 r2cloud java[1064]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 25 13:25:57 r2cloud java[1064]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 25 13:25:57 r2cloud java[1064]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 13:26:31 r2cloud java[1064]: UncaughtException at: NanoHttpd Request Processor (#6)
Sep 25 13:26:31 r2cloud java[1064]: java.lang.OutOfMemoryError: Java heap space
Sep 25 13:26:31 r2cloud java[1064]:         at ru.r2cloud.jradio.BeaconInputStream.readProtocolv1(BeaconInputStream.java:44)
Sep 25 13:26:31 r2cloud java[1064]:         at ru.r2cloud.jradio.BeaconInputStream.hasNext(BeaconInputStream.java:27)
Sep 25 13:26:31 r2cloud java[1064]:         at ru.r2cloud.web.api.observation.ObservationLoad.convertBeacons(ObservationLoad.java:93)
Sep 25 13:26:31 r2cloud java[1064]:         at ru.r2cloud.web.api.observation.ObservationLoad.doGet(ObservationLoad.java:68)
Sep 25 13:26:31 r2cloud java[1064]:         at ru.r2cloud.web.WebServer.serve(WebServer.java:77)
Sep 25 13:26:31 r2cloud java[1064]:         at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:945)
Sep 25 13:26:31 r2cloud java[1064]:         at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
Sep 25 13:26:31 r2cloud java[1064]:         at java.base/java.lang.Thread.run(Thread.java:834)
Sep 25 13:26:31 r2cloud java[1064]:  [ru.r2cloud.R2Cloud$2 uncaughtException]
Sep 25 13:26:31 r2cloud java[1064]: UncaughtException at: NanoHttpd Request Processor (#7)
Sep 25 13:26:31 r2cloud java[1064]: java.lang.OutOfMemoryError: Java heap space
Sep 25 13:26:31 r2cloud java[1064]:  [ru.r2cloud.R2Cloud$2 uncaughtException]

dernasherbrezon, 2022-09-25:
Can you also remove cached satellites from:

```
rm /home/pi/r2cloud/data/leosatdata.json
```

ArgoNavi, 2022-09-25:
That did it, everything works now and I can view my images :grinning: