dernasherbrezon/r2cloud #132: TLE not updating

RcamiruagaJ, 2022-09-30:
Hi Andrey,
Following our chat in Telegram related, I notice that TLE is not update
I attached the last log (200 lines)
Have a nice day
Ramon
[JournalLast200.pdf](https://github.com/dernasherbrezon/r2cloud/files/9684154/JournalLast200.pdf)

dernasherbrezon, 2022-09-30:
Looks according to logs:

```
sep 30 08:11:57 raspberrypi java[425]: loading tle from: http://mstl.atl.calpol>
sep 30 08:11:58 raspberrypi java[425]: received tle for 136 satellites [ru.r2cl>
sep 30 08:11:58 raspberrypi java[425]: loading tle from: http://celestrak.com/N>
sep 30 08:12:00 raspberrypi java[425]: received tle for 253 satellites [ru.r2cl>
sep 30 08:12:00 raspberrypi java[425]: loading tle from: http://celestrak.com/N>
sep 30 08:12:02 raspberrypi java[425]: received tle for 6603 satellites [ru.r2c>
```

Where do you see the TLE error?

RcamiruagaJ, 2022-09-30:
> Looks good according to logs:

In the same software
Please open attached file in your browser.

Really according journal looks good....¿¿¿¿?????

dernasherbrezon, 2022-10-01:
Hi,

Yes, logs are good. The attached file is not .pdf, but just a plain text file.

closed 2022-10-01 by dernasherbrezon

RcamiruagaJ, 2022-10-03:
> Why the first line in TLE page is not refresh, in spite off TLE for
> all satellites are updated.???
>
> Than you for your help
>
> Ramón

Hi Andrey,

Following our subject.

I can not fix the issue... I am thinking that could be a problem of 
sincronization time between r2cloud soft and RPi OS.

I have tried change time of RPi in configuration fixed to UTC time, not 
to my location (Chile is UTC+3), but problem is not resolved.

The page of TLE (r2cloud soft), is glued indicating a previous date 
(instead of the journal show that time observations are in UTC time).

Before two months ago I have not this problem.

I would appreciate your comments.

Have a nice day

Ramon

dernasherbrezon, 2022-10-03:
Fixed in 415edad301a88c4f3ac9c5a18fcc2f55df3babe5. TLE last update time was taken from props, not from the TLE cache. Was affecting only representation on UI rather than actual schedule.

RcamiruagaJ, 2022-10-11:
Why the first line in TLE page is not refresh, in spite off TLE for all 
satellites are updated.???

Than you for your help

Ramón