dernasherbrezon/r2cloud #135: Number of times you download images.

TenerifeHR, 2022-11-13:
Hi I would like to know why it only downloads images from the nooa or the meteor, 2 times a day, once in the morning and once at night, the satellite passes over me more times a day broadcasting image, Thank you.

dernasherbrezon, 2022-11-15:
Hi, schedules are based on the elevation. By default r2cloud schedule observations for minimum elevation 8 degrees and guaranteed 20 degrees. If your antenna / setup works well on lower elevations, then you can change these settings in the configuration to be something smaller. I.e. 0 degrees both.

closed 2022-11-15 by dernasherbrezon

TenerifeHR, 2022-11-15:
Thank you very much I will try to find out how to do it, I hope to be able to collect images more times a day.