dernasherbrezon/r2cloud #136: Decoding observations accumulating

ArgoNavi, 2022-12-11:
Over the past few days I've been having a problem with observations taking hours (6+) to decode, the result is an unending accumulation of observations that are decoding.

Currently the only way to fix this is to disable observations and wait several hours (8+) for everything to finish, however once enabled again the accumulation begins.

METEOR-M 2 is disabled.

[image]

ArgoNavi, 2022-12-11:
Here are the Logs

```shell
-- Logs begin at Thu 2019-02-14 05:11:59 EST, end at Sun 2022-12-11 18:44:08 EST. --
Dec 11 17:08:00 r2cloud java[18571]: unable to save. dest already exist: /home/pi/r2cloud/./data/satellites/39430/data/1670777717179-39430-0/data.bin [ru.r2cloud.satellite.ObservationDao saveData]
Dec 11 17:08:00 r2cloud java[18571]: [1670777717179-39430-0] uploading observation [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Dec 11 17:08:01 r2cloud java[18571]: stopping: rtl_sdr for 1670795909835-46912-0 [ru.r2cloud.util.Util shutdown]
Dec 11 17:08:01 r2cloud java[18571]: Signal caught, exiting!
Dec 11 17:08:01 r2cloud java[18571]: [1670777717179-39430-0] observation uploaded [ru.r2cloud.cloud.LeoSatDataService uploadObservation]
Dec 11 17:08:01 r2cloud java[18571]: [1670774247643-51044-0] decoding [ru.r2cloud.satellite.decoder.DecoderService runInternally]
Dec 11 17:08:01 r2cloud java[18571]: [1670795909835-46912-0] observation time passed. cancelling rotation [ru.r2cloud.satellite.RotatorService$1 run]
Dec 11 17:08:02 r2cloud java[18571]: User cancel, exiting...
Dec 11 17:08:02 r2cloud java[18571]: [1670795909835-46912-0] rtl_sdr stopped: 143 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Dec 11 17:08:02 r2cloud java[18571]: deleting old observation: ./data/satellites/52191/data/1670518345760-52191-0 last update time: Thu Dec 08 17:01:15 UTC 2022 [ru.r2cloud.satellite.TimeSizeRetention ind
Dec 11 17:08:02 r2cloud java[18571]: deleting old observation: ./data/satellites/43132/data/1670518981461-43132-0 last update time: Thu Dec 08 17:10:31 UTC 2022 [ru.r2cloud.satellite.TimeSizeRetention ind
Dec 11 17:17:45 r2cloud java[18571]: started with arguments: ./etc/rtl_sdr_wrapper.sh -rtl rtl_sdr -f 437462989 -d 0 -s 240000 -g 40.0 -p 0 -o /tmp/40014-1670797065797-40014-0.raw.gz [ru.r2cloud.util.Proc
Dec 11 17:17:45 r2cloud java[18571]: [1670797065797-40014-0] moving rotator to Position [azimuth=123.31903499881038, elevation=5.0000369676605105] [ru.r2cloud.satellite.RotatorService$1 run]
Dec 11 17:17:45 r2cloud java[18571]: Found 1 device(s):
Dec 11 17:17:45 r2cloud java[18571]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Dec 11 17:17:45 r2cloud java[18571]: Using device 0: Generic RTL2832U OEM
Dec 11 17:17:46 r2cloud java[18571]: Found Rafael Micro R820T tuner
Dec 11 17:17:46 r2cloud java[18571]: [R82XX] PLL not locked!
Dec 11 17:17:46 r2cloud java[18571]: Sampling at 240000 S/s.
Dec 11 17:17:46 r2cloud java[18571]: Tuned to 437462989 Hz.
Dec 11 17:17:46 r2cloud java[18571]: Tuner gain set to 40.20 dB.
Dec 11 17:17:46 r2cloud java[18571]: Reading samples in async mode...
Dec 11 17:23:13 r2cloud java[18571]: loading satellites from leosatdata [ru.r2cloud.cloud.LeoSatDataClient loadSatellites]
Dec 11 17:23:14 r2cloud java[18571]: no satellite updates from leosatdata [ru.r2cloud.cloud.LeoSatDataClient loadSatellites]
Dec 11 17:23:14 r2cloud java[18571]: loading tle from: http://mstl.atl.calpoly.edu/~ops/keps/kepler.txt [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:15 r2cloud java[18571]: received tle for 136 satellites [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:15 r2cloud java[18571]: loading tle from: http://celestrak.com/NORAD/elements/satnogs.txt [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:15 r2cloud java[18571]: received tle for 252 satellites [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:15 r2cloud java[18571]: loading tle from: http://celestrak.com/NORAD/elements/active.txt [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:16 r2cloud java[18571]: received tle for 6847 satellites [ru.r2cloud.tle.CelestrakClient downloadTle]
Dec 11 17:23:16 r2cloud java[18571]: check for not processed observations [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670784052694-40042-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670792393555-39444-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670784627898-46914-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670789782198-46912-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670780702937-30776-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670794241800-43197-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670786458317-30776-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670792247245-30776-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670795909835-46912-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670791605184-44419-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
Dec 11 17:23:16 r2cloud java[18571]: resuming decoding: 1670779377193-43743-0 [ru.r2cloud.satellite.decoder.DecoderService retryObservations]
[... cut ...]
Dec 11 18:40:23 r2cloud java[18571]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Dec 11 18:40:23 r2cloud java[18571]: Using device 0: Generic RTL2832U OEM
Dec 11 18:40:24 r2cloud java[18571]: Found Rafael Micro R820T tuner
Dec 11 18:40:24 r2cloud java[18571]: [R82XX] PLL not locked!
Dec 11 18:40:24 r2cloud java[18571]: Sampling at 240000 S/s.
Dec 11 18:40:24 r2cloud java[18571]: Tuned to 401047880 Hz.
Dec 11 18:40:24 r2cloud java[18571]: Tuner gain set to 40.20 dB.
Dec 11 18:40:24 r2cloud java[18571]: Reading samples in async mode...

dernasherbrezon, 2022-12-12:
It looks non-processed observations scheduled twice. Will work on a fix.

dernasherbrezon, 2022-12-12:
Should be fixed in 5d92a9b78d0ab4b04eb8f8568e908cf52a0c2382

closed 2022-12-12 by dernasherbrezon