dernasherbrezon/r2cloud #137: Installation fail

jtagris, 2023-03-02:
Hello, thank you for your work on this project, this is just what I needed.

but after many tries I can't access the GUI, every time i try to access https://raspberrypi.local I get the error "502" R2cloud is not running.

I'm on pi4, with ethernet, new sd card.

if you may help me I be grateful

Kind regards 
Jtagris

dernasherbrezon, 2023-03-02:
Hey,

Can you login to raspberry PI and check check logs?

```
journalctl -u r2cloud -n 200
```

Recently we had an issue with JDK upgrade.

jtagris, 2023-03-03:
the problem came with the image for raspberry, installing raspi-os and fetching the binaries it works!

unfortunately I no longer have access to the logs of the machine but I believe I have seen problems with jdk on the terminal

I attach the logs of my new installation, everything does not seem correct and I have no observations,do you see something wrong I am sorry for my pitiful English and IT skills 

Have a good day 

end: Sun Mar 05 00:23:20 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43880-0. start: Sun Mar 05 00:24:26 UTC 2023 end: Sun Mar 05 00:30:24 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53379-0. start: Sun Mar 05 00:30:24 UTC 2023 end: Sun Mar 05 00:37:37 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53374-0. start: Sun Mar 05 00:39:01 UTC 2023 end: Sun Mar 05 00:45:51 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53373-0. start: Sun Mar 05 00:45:51 UTC 2023 end: Sun Mar 05 00:49:39 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[806]: usb_open error -3
Mar 03 14:03:49 raspberrypi java[806]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53378-0. start: Sun Mar 05 00:52:13 UTC 2023 end: Sun Mar 05 00:58:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[806]: Failed to open rtlsdr device #0.
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 52148-0. start: Sun Mar 05 01:00:47 UTC 2023 end: Sun Mar 05 01:05:54 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55181-0. start: Sun Mar 05 01:06:33 UTC 2023 end: Sun Mar 05 01:13:15 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 48928-0. start: Sun Mar 05 01:13:42 UTC 2023 end: Sun Mar 05 01:20:27 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39428-0. start: Sun Mar 05 01:26:06 UTC 2023 end: Sun Mar 05 01:35:10 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [1677848601115-46489-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55129-0. start: Sun Mar 05 01:45:57 UTC 2023 end: Sun Mar 05 01:53:05 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40042-0. start: Sun Mar 05 01:54:05 UTC 2023 end: Sun Mar 05 02:02:13 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55124-1. start: Sun Mar 05 02:03:12 UTC 2023 end: Sun Mar 05 02:09:41 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 54684-0. start: Sun Mar 05 02:15:04 UTC 2023 end: Sun Mar 05 02:23:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40378-0. start: Sun Mar 05 02:28:18 UTC 2023 end: Sun Mar 05 02:34:27 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 46495-0. start: Sun Mar 05 02:35:18 UTC 2023 end: Sun Mar 05 02:43:45 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 48928-0. start: Sun Mar 05 02:46:57 UTC 2023 end: Sun Mar 05 02:54:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39444-0. start: Sun Mar 05 02:58:28 UTC 2023 end: Sun Mar 05 03:05:59 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 52899-0. start: Sun Mar 05 03:06:13 UTC 2023 end: Sun Mar 05 03:16:29 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39440-0. start: Sun Mar 05 03:20:50 UTC 2023 end: Sun Mar 05 03:30:37 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40042-0. start: Sun Mar 05 03:30:37 UTC 2023 end: Sun Mar 05 03:37:45 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43192-0. start: Sun Mar 05 03:38:58 UTC 2023 end: Sun Mar 05 03:44:32 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53106-0. start: Sun Mar 05 03:44:32 UTC 2023 end: Sun Mar 05 03:43:31 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43197-0. start: Sun Mar 05 03:47:59 UTC 2023 end: Sun Mar 05 03:54:52 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 44406-0. start: Sun Mar 05 03:55:57 UTC 2023 end: Sun Mar 05 04:03:15 UTC 2023 [ru.r2cloud.device.Device schedule]
[... cut ...]
Mar 03 14:08:45 raspberrypi java[873]: usb_open error -3
Mar 03 14:08:45 raspberrypi java[873]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Mar 03 14:08:45 raspberrypi java[873]: Failed to open rtlsdr device #0.
Mar 03 14:08:45 raspberrypi java[389]: [1677848925476-39428-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Mar 03 14:13:40 raspberrypi java[389]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]

Lukas0025, 2023-03-03:
Something is bad with your SDR device. usb_open error.

try install the udev rules file rtl-sdr.rules

Odesláno ze zařízení s Windows 10

Od: jtagris
Odesláno:pátek 3. března 2023 14:17
Komu: dernasherbrezon/r2cloud
Kopie: Subscribed
Předmět: Re: [dernasherbrezon/r2cloud] Installation fail (Issue #137)

the problem came with the image for raspberry, installing raspi-os and fetching the binaries it works!
unfortunately I no longer have access to the logs of the machine but I believe I have seen problems with jdk on the terminal
I attach the logs of my new installation, everything does not seem correct and I have no observations,do you see something wrong I am sorry for my pitiful English and IT skills
Have a good day
end: Sun Mar 05 00:23:20 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43880-0. start: Sun Mar 05 00:24:26 UTC 2023 end: Sun Mar 05 00:30:24 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53379-0. start: Sun Mar 05 00:30:24 UTC 2023 end: Sun Mar 05 00:37:37 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53374-0. start: Sun Mar 05 00:39:01 UTC 2023 end: Sun Mar 05 00:45:51 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53373-0. start: Sun Mar 05 00:45:51 UTC 2023 end: Sun Mar 05 00:49:39 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[806]: usb_open error -3
Mar 03 14:03:49 raspberrypi java[806]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53378-0. start: Sun Mar 05 00:52:13 UTC 2023 end: Sun Mar 05 00:58:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[806]: Failed to open rtlsdr device #0.
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 52148-0. start: Sun Mar 05 01:00:47 UTC 2023 end: Sun Mar 05 01:05:54 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55181-0. start: Sun Mar 05 01:06:33 UTC 2023 end: Sun Mar 05 01:13:15 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 48928-0. start: Sun Mar 05 01:13:42 UTC 2023 end: Sun Mar 05 01:20:27 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39428-0. start: Sun Mar 05 01:26:06 UTC 2023 end: Sun Mar 05 01:35:10 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [1677848601115-46489-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55129-0. start: Sun Mar 05 01:45:57 UTC 2023 end: Sun Mar 05 01:53:05 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40042-0. start: Sun Mar 05 01:54:05 UTC 2023 end: Sun Mar 05 02:02:13 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 55124-1. start: Sun Mar 05 02:03:12 UTC 2023 end: Sun Mar 05 02:09:41 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 54684-0. start: Sun Mar 05 02:15:04 UTC 2023 end: Sun Mar 05 02:23:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40378-0. start: Sun Mar 05 02:28:18 UTC 2023 end: Sun Mar 05 02:34:27 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 46495-0. start: Sun Mar 05 02:35:18 UTC 2023 end: Sun Mar 05 02:43:45 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 48928-0. start: Sun Mar 05 02:46:57 UTC 2023 end: Sun Mar 05 02:54:28 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39444-0. start: Sun Mar 05 02:58:28 UTC 2023 end: Sun Mar 05 03:05:59 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 52899-0. start: Sun Mar 05 03:06:13 UTC 2023 end: Sun Mar 05 03:16:29 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 39440-0. start: Sun Mar 05 03:20:50 UTC 2023 end: Sun Mar 05 03:30:37 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 40042-0. start: Sun Mar 05 03:30:37 UTC 2023 end: Sun Mar 05 03:37:45 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43192-0. start: Sun Mar 05 03:38:58 UTC 2023 end: Sun Mar 05 03:44:32 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 53106-0. start: Sun Mar 05 03:44:32 UTC 2023 end: Sun Mar 05 03:43:31 UTC 2023 [ru.r2cloud.device.Device schedule]
Mar 03 14:03:49 raspberrypi java[389]: [sdr-0] scheduled next pass for 43197-0. start: Sun Mar 05 03:47:59 UTC 2023 end: Sun Mar 05 03:54:52 UTC 2023 [ru.r2cloud.device.Device schedule]
[... cut ...]
Mar 03 14:08:45 raspberrypi java[873]: usb_open error -3
Mar 03 14:08:45 raspberrypi java[873]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Mar 03 14:08:45 raspberrypi java[873]: Failed to open rtlsdr device #0.
Mar 03 14:08:45 raspberrypi java[389]: [1677848925476-39428-0] invalid response code rtl_sdr: 1 [ru.r2cloud.satellite.reader.RtlSdrReader start]
Mar 03 14:13:40 raspberrypi java[389]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]

closed 2023-03-03 by dernasherbrezon

dernasherbrezon, 2023-03-03:
This is something to do with your disk. /dev/sda2.

jtagris, 2023-03-03:
get out my ssd and Flashed it on other sd card and finally everything is working fine 
thanks for your time