dernasherbrezon/r2cloud #138: Build up of observations & SSH login details

TechMindsYT, 2023-05-20:
Hi all,

I have a new install  which has been working okay until today. Now the observations are just saying "decoding", around 30 of them.

I want to SSH into the pi to try to find the logs, but I the standard pi / raspberry does not work.

I am using the latest R2Cloud image downloaded from Github.

Thanks

dernasherbrezon, 2023-05-20:
Hi,

I think ssh login by default is disable on all RPi images. r2cloud image is based on the official hence SSH disabled also. Can you turn on SSH-server if you mount SD card into your PC and create a file named "ssh" there, unmount and insert back into RPi. Can't find the recent guide about it, but it always worked. 

Once on the host, you can check logs using the command ```journalctl -u r2cloud -n 200``` or ```journalctl -u r2cloud -n 800```.

TechMindsYT, 2023-05-20:
> Hi,
> 
> I think ssh login by default is disabled on all RPi images. r2cloud image is based on the official hence SSH disabled also. You can turn on SSH-server if you:
> 
> * mount SD card into your PC
> * create a file named "ssh" there
> * unmount
> * insert back into RPi.
> 
> Can't find the recent guide about it, but it always worked.
> 
> Once on the host, you can check logs using the command `journalctl -u r2cloud -n 200` or `journalctl -u r2cloud -n 800`.

Well, that's the thing, I have done that to gain access via SSH using Putty, but when presented with the username and password the default "pi" and "raspberry" do not appear to work. So, I was wondering if someone know what it is?

Thanks

dernasherbrezon, 2023-05-20:
r2cloud image don't have anything specific to SSH. Here is the repo btw: https://github.com/dernasherbrezon/pi-gen

TechMindsYT, 2023-05-21:
How do I fix this if I cannot SSH?

I want to promote this useful application on my YouTube channel, but I cant unless I can fix this.

[image]

dernasherbrezon, 2023-05-21:
Hi, it seems the issue is related to https://github.com/dernasherbrezon/r2cloud/issues/136 . It was fixed in December 2022. Are you using the latest image? Without logs it's really hard to troubleshoot. 

There was some update in recent RPi images. Can you try:

```
There are also mechanisms to preconfigure an image without using Imager. To set up a user on first boot and bypass the wizard completely, create a file called userconf or userconf.txt in the boot partition of the SD card; this is the part of the SD card which can be seen when it is mounted in a Windows or MacOS computer.
```

More details at https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

TechMindsYT, 2023-05-21:
is there a way to access the logs?

To confirm I am using the latest pre-built image downloaded from this github.

dernasherbrezon, 2023-05-21:
No, logs might contain secure information, thus no access without username and password :)

Let me test SSH access on the latest image.

dernasherbrezon, 2023-05-21:
Yes, they removed default username and password from images. To use pi you need:

1. Mount SD card into PC
2. Create a file userconf.txt there
3. Put ```pi:$6$XVh7/PVvihco7hhw$NwS6PHSyf3.MUvd6kdS7l997YpQNAXhsmjR2rQ8zMCqJMAdrjirbXpigFHx/ytx3huiOMxI21xfBjAEOTD3QU/``` 
4. Login via SSH using username "pi" and old password.

Once logged in, can you please check logs and attach here?

TechMindsYT, 2023-05-21:
Hi, thanks, I can now access the Pi via SSH.

However, where are the logs located? Could you please provide the path and filename to the log file please?

Thanks

dernasherbrezon, 2023-05-22:
Once on the host, you can check logs using the command ```journalctl -u r2cloud -n 200``` or ```journalctl -u r2cloud -n 800```. More troubleshooting tips in the documentation: https://github.com/dernasherbrezon/r2cloud/wiki/Troubleshooting-guide

dernasherbrezon, 2023-06-02:
Do you still see the issue? You can join telegram channel where we can troubleshoot the issue live: https://t.me/leosatdata

closed 2023-06-23 by dernasherbrezon