dernasherbrezon/r2cloud #142: r2 server not running error

radiorusso, 2024-07-31:
I had this running last year now when it starts it says r2cloud not running also what are the user names and password to login to the console ? tried pi as the user name and raspberry as the password  . I would like to get this thing running again as it is hurricane season

thanks in advance   Russ

dernasherbrezon, 2024-08-01:
Hi,

r2cloud's image is based on standard Raspbian OS. So all guides, tutorials and troubleshooting are valid. Here is some useful links to fix SSH login:
https://www.raspberrypi.com/documentation/computers/remote-access.html#ssh
https://raspberrypi.stackexchange.com/questions/142990/default-password-not-working-with-ssh
https://forums.raspberrypi.com/viewtopic.php?t=361070

Once logged in, follow common troubleshooting steps:
https://github.com/dernasherbrezon/r2cloud/wiki/Troubleshooting-guide

radiorusso, 2024-08-03:
SO I can see that on the pi  the high performance web server and reverse proxy server load correctly and are running  so IS this an issue on  a server outside the pi ?  could this be a failure to connect to the internet ??
Thanks russ

dernasherbrezon, 2024-08-03:
Hi,

Can you check logs according to https://github.com/dernasherbrezon/r2cloud/wiki/Troubleshooting-guide ? It should output any errors if r2cloud failed to start or not available.

closed 2024-08-21 by dernasherbrezon

radiorusso, 2024-08-21:
Sorry non of that shed light on the issue   sorry moved on to a different approach. It was a good program but just wont run. You suggestions didnt help