dernasherbrezon/r2cloud #146: Nooelec NESDR SMArTee v2 not working

Nammurg, 2024-09-27:
Hello, as the title says, I have a Nooelec SMArTee v2 dongle based on RTL2832U chipset.
I run the latest Pi image available of r2cloud. Pi is 3B+ model.
Within the console of the Pi, I can run rtl_test, rtl_eeprom, rtl_fm and they recognise the tuner just fine. 
However, within r2cloud devices menu, it shows up with a strange name and no serial number...
In addition, I am not able to record any satellite passes even though they are being scheduled...
Any help would be much appreciated!

------------------------------------------------------------------------------------------------------------

**r2cloud@r2cloud:~ $ rtl_test** 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: AMZN20241026

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
User cancel, exiting...
Samples per million lost (minimum): 0

------------------------------------------------------------------------------------------------------------

**r2cloud@r2cloud:~ $ rtl_eeprom** 
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		AMZN20241026
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

------------------------------------------------------------------------------------------------------------

**r2cloud@r2cloud:~ $ rtl_fm -f 137912500 -d 0 -s 60000 -g 0.0 -p 1 -E deemp -F 9**
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: AMZN20241026

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 0.00 dB.
Tuner error set to 1 ppm.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.

------------------------------------------------------------------------------------------------------------

**r2cloud@r2cloud:~ $ journalctl -u r2cloud -n 200**
Sep 27 21:51:42 r2cloud r2cloud[1634]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Sep 27 21:51:42 r2cloud r2cloud[1634]: Failed to open rtlsdr device #0.
Sep 27 21:51:42 r2cloud r2cloud[430]: [1727470301889-28654-0] invalid response code rtl_fm: 1 [ru.r2cloud.satellite.reader.RtlFmReader startInternally]
[... 218 lines cut ...]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 08:55:51 UTC 2024 end: Sun Sep 29 09:05:16 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Sun Sep 29 09:42:17 UTC 2024 end: Sun Sep 29 09:55:00 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 10:35:04 UTC 2024 end: Sun Sep 29 10:50:04 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Sun Sep 29 11:23:23 UTC 2024 end: Sun Sep 29 11:27:06 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Sun Sep 29 11:27:06 UTC 2024 end: Sun Sep 29 11:41:25 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 12:15:52 UTC 2024 end: Sun Sep 29 12:30:46 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Sun Sep 29 12:48:40 UTC 2024 end: Sun Sep 29 12:54:44 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Sun Sep 29 13:08:34 UTC 2024 end: Sun Sep 29 13:17:37 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 13:57:13 UTC 2024 end: Sun Sep 29 14:07:50 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Sun Sep 29 14:23:17 UTC 2024 end: Sun Sep 29 14:36:11 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Sun Sep 29 16:11:03 UTC 2024 end: Sun Sep 29 16:21:55 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Sun Sep 29 17:43:16 UTC 2024 end: Sun Sep 29 17:56:39 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Sun Sep 29 18:00:35 UTC 2024 end: Sun Sep 29 18:11:13 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 18:51:02 UTC 2024 end: Sun Sep 29 18:59:36 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Sun Sep 29 19:27:41 UTC 2024 end: Sun Sep 29 19:42:10 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Sun Sep 29 20:27:03 UTC 2024 end: Sun Sep 29 20:41:08 UTC 2024 [ru.r2cloud.device.Device schedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: observations scheduled. next update at: Sun Sep 29 21:07:05 UTC 2024 [ru.r2cloud.device.DeviceManager reschedule]
Sep 27 22:07:05 r2cloud r2cloud[430]: webserver is listening on localhost:8097 [ru.r2cloud.web.WebServer start]
Sep 27 22:07:05 r2cloud r2cloud[430]: ================================= [ru.r2cloud.R2Cloud start]
Sep 27 22:07:05 r2cloud r2cloud[430]: =========== started ============= [ru.r2cloud.R2Cloud start]
Sep 27 22:07:05 r2cloud r2cloud[430]: ================================= [ru.r2cloud.R2Cloud start]
Sep 27 22:07:11 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 27 22:07:35 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 27 22:08:30 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 27 22:15:30 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]

dernasherbrezon, 2024-09-27:
Hi,

It might be some issue with permissions. You're running commands from r2cloud user, while r2cloud application is running under "pi" user. Can you check if pi user has enough permissions? I recall it should be part of "wheel" group.

Nammurg, 2024-09-28:
Hi, thank you for the swift response!

Interesting, the user PI does not exist anymore as they are forcing you now to create a custom username/password...

https://www.tomshardware.com/news/raspberry-pi-default-username-removed

dernasherbrezon, 2024-09-29:
Is it working now?

Nammurg, 2024-09-29:
No it is not as nothing changed, as the Pi user does not exist anymore...

I will try to install it in an older version of raspbian and see if that fixes it. 

Another behaviour I we experienced when using the latest image was that as soon as I setup the new user and password r2cloud eould stop working altogether and I could not even get the web interface ...

dernasherbrezon, 2024-09-29:
r2cloud .deb file auto-create "pi" user. In the previous versions of image "pi" user already existed with all required permissions. And this step was simply no-op. However newer versions don't have "pi" user and auto-creation scripts auto-create "pi" with some missing permissions. I will closely compare permissions and release new image.

Nammurg, 2024-09-29:
I am here if you need anything from my side and if I can help with any testing, etc. I am more than happy to help!

Also want to take the opportunity to say a big THANK YOU for this amazing project!

dernasherbrezon, 2024-09-29:
Fixed in b8c0111e91409a030a5f4546ad88100e5f2344a6. Should be available in the next version:

```
sudo apt update
sudo apt install r2cloud
```

Can you try?

Nammurg, 2024-09-29:
> Fixed in [b8c0111](https://github.com/dernasherbrezon/r2cloud/commit/b8c0111e91409a030a5f4546ad88100e5f2344a6). Should be available in the next version:
> 
> ```
> sudo apt update
> sudo apt install r2cloud
> ```
> 
> Can you try?

I run both commands and the result is: 
r2cloud is already the newest version (20240922110805).

dernasherbrezon, 2024-09-29:
The new version will be slowly distributed to all Amazon caches in the next day.

Nammurg, 2024-09-30:
Good morning, I have managed to update r2cloud and thinks look very promising!
The Device is now recognised correctly and I am able to "lock it down" based on it's serial number.
I will keep it online for couple of satellite passes to see if it is now able to schedule jobs and I will report back here again.
Thank you very much for the swift response! :)

Nammurg, 2024-09-30:
OK so schedules are going through, however, I get 0 frames and a blank spectrogram... 
Any Ideas?

dernasherbrezon, 2024-09-30:
Getting 0 frames is ok. Sometimes signal is too weak to be decoded. Blank spectrogram is an issue. Can you please check logs?

Nammurg, 2024-09-30:
r2cloud@r2cloud:~ $ journalctl -u r2cloud -n 200
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Tue Oct 01 20:19:08 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Tue Oct 01 20:54:09 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Tue Oct 01 21:42:26 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Tue Oct 01 22:38:16 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Tue Oct 01 23:24:52 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Wed Oct 02 03:35:33 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Wed Oct 02 05:14:14 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Wed Oct 02 06:46:04 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Wed Oct 02 06:57:25 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Wed Oct 02 08:25:35 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Wed Oct 02 09:10:18 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Wed Oct 02 09:58:53 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 25338-0. start: Wed Oct 02 10:06:29 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Wed Oct 02 10:50:57 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Wed Oct 02 11:39:06 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 33591-0. start: Wed Oct 02 12:32:52 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 28654-0. start: Wed Oct 02 13:20:38 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: [rtlsdr.0] scheduled next pass for 40069-0. start: Wed Oct 02 15:02:44 UTC 2024 e>
Sep 30 17:37:54 r2cloud r2cloud[430]: observations scheduled. next update at: Wed Oct 02 16:37:54 UTC 2024 [ru.r2cloud.>
Sep 30 17:37:54 r2cloud r2cloud[430]: webserver is listening on localhost:8097 [ru.r2cloud.web.WebServer start]
Sep 30 17:37:54 r2cloud r2cloud[430]: ================================= [ru.r2cloud.R2Cloud start]
Sep 30 17:37:54 r2cloud r2cloud[430]: =========== started ============= [ru.r2cloud.R2Cloud start]
Sep 30 17:37:54 r2cloud r2cloud[430]: ================================= [ru.r2cloud.R2Cloud start]
Sep 30 17:54:15 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 30 17:54:24 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 30 17:54:27 r2cloud r2cloud[430]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 30 17:54:56 r2cloud systemd[1]: Stopping r2cloud.service - R2Cloud Service...
Sep 30 17:54:56 r2cloud r2cloud[430]: stopping [ru.r2cloud.R2Cloud grafullyShutdown]
Sep 30 17:54:56 r2cloud r2cloud[430]: [rtlsdr.0] device stopped [ru.r2cloud.device.Device stop]
Sep 30 17:54:56 r2cloud r2cloud[430]: =========== stopped ============= [ru.r2cloud.R2Cloud grafullyShutdown]
Sep 30 17:54:57 r2cloud systemd[1]: r2cloud.service: Deactivated successfully.
Sep 30 17:54:57 r2cloud systemd[1]: Stopped r2cloud.service - R2Cloud Service.
Sep 30 17:54:57 r2cloud systemd[1]: r2cloud.service: Consumed 31.808s CPU time.
-- Boot 761792d3508f4da6810a4d8c56c213f7 --
Sep 30 17:55:01 r2cloud systemd[1]: Started r2cloud.service - R2Cloud Service.
Sep 30 17:55:08 r2cloud r2cloud[436]: retention: keep last 10240Mb of observations and 10 of raw IQ data [ru.r2cloud.sa>
Sep 30 17:55:08 r2cloud r2cloud[436]: some influxdb parameters not configured. skip sending metrics [ru.r2cloud.cloud.I>
Sep 30 17:55:13 r2cloud r2cloud[436]: HIGH: 0 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 30 17:55:13 r2cloud r2cloud[436]: NORMAL: 252 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 30 17:55:13 r2cloud r2cloud[436]:   CONFIG: 252 [ru.r2cloud.satellite.SatelliteDao printStatsBySource]
Sep 30 17:55:13 r2cloud r2cloud[436]: started with arguments: [rtl_test, -t] [ru.r2cloud.util.ProcessFactory create]
Sep 30 17:55:14 r2cloud r2cloud[436]: Skip TLE update. Last update was Sun Sep 29 21:14:22 UTC 2024. Next update: Tue O>
Sep 30 17:55:14 r2cloud r2cloud[436]: check for not processed observations [ru.r2cloud.satellite.decoder.DecoderService>
Sep 30 17:55:14 r2cloud r2cloud[436]: [rtlsdr.0] device started [ru.r2cloud.device.Device start]
Sep 30 17:55:17 r2cloud r2cloud[436]: HIGH: satellites 0 observations 0 [ru.r2cloud.satellite.Schedule scheduleSatellit>
Sep 30 17:55:17 r2cloud r2cloud[436]: NORMAL: satellites 0 observations 0 [ru.r2cloud.satellite.Schedule scheduleSatell>
Sep 30 17:55:17 r2cloud r2cloud[436]: NORMAL: satellites 4 observations 41 [ru.r2cloud.satellite.Schedule scheduleSatel>
Sep 30 17:55:17 r2cloud r2cloud[436]: [rtlsdr.0] scheduled next pass for 25338-0. start: Mon Sep 30 17:23:41 UTC 2024 e>
Sep 30 17:55:17 r2cloud r2cloud[436]: [rtlsdr.0] scheduled next pass for 25338-0. start: Mon Sep 30 19:02:42 UTC 2024 e>
[... cut ...]
Sep 30 20:27:08 r2cloud r2cloud[2902]: Found Rafael Micro R820T tuner
Sep 30 20:27:08 r2cloud r2cloud[2902]: Tuner gain set to 0.00 dB.
Sep 30 20:27:08 r2cloud r2cloud[2902]: Tuned to 137580000 Hz.
Sep 30 20:27:08 r2cloud r2cloud[2902]: Oversampling input by: 32x.
Sep 30 20:27:08 r2cloud r2cloud[2902]: Oversampling output by: 1x.
Sep 30 20:27:08 r2cloud r2cloud[2902]: Buffer size: 4.27ms
Sep 30 20:27:08 r2cloud r2cloud[2902]: Sampling at 1920000 S/s.
Sep 30 20:27:08 r2cloud r2cloud[2902]: Output at 60000 Hz.

dernasherbrezon, 2024-09-30:
You need to install wxtoimg to decode NOAA weather images. The latest ARM64 architecture is not supported by wxtoimg, so this dependency is optional:

```
apt install wxtoimg
```

Nammurg, 2024-09-30:
Thank you, I have installed it now.
Is there any way I can process the data captured by running a command?

Nammurg, 2024-10-01:
I have tried to install r2cloud on a brand new 32bit debian bullseye for the PI 3B+ but I am getting the below error when trying to install r2cloud.

r2cloud@r2cloud:~ $ sudo apt-get install r2cloud
Get:1 http://apt.leosatdata.com bullseye/main armhf r2cloud-jdk armhf 17.0.5-3 [45.7 MB]
Err:1 http://apt.leosatdata.com bullseye/main armhf r2cloud-jdk armhf 17.0.5-3
  File has unexpected size (45715780 != 45723774). Mirror sync in progress? [IP: 18.245.187.97 80]
  Hashes of expected file:
   - SHA256:7058b614d2ca7a09747a8ec722ea1a5d48b61e7ecd67ea3b6ca58397e6dc5602
   - SHA1:021e4b5edb66ff9fff55c6b6c93a60bda441ceb0 [weak]
   - MD5Sum:c3486ca57afe5d0ff03e67c1613742c7 [weak]
   - Filesize:45723774 [weak]
E: Failed to fetch http://apt.leosatdata.com/pool/main/r/r2cloud-jdk/r2cloud-jdk_17.0.5-3_armhf.deb  File has unexpected size (45715780 != 45723774). Mirror sync in progress? [IP: 18.245.187.97 80]
   Hashes of expected file:
    - SHA256:7058b614d2ca7a09747a8ec722ea1a5d48b61e7ecd67ea3b6ca58397e6dc5602
    - SHA1:021e4b5edb66ff9fff55c6b6c93a60bda441ceb0 [weak]
    - MD5Sum:c3486ca57afe5d0ff03e67c1613742c7 [weak]
    - Filesize:45723774 [weak]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

----

r2cloud@r2cloud:~ $ sudo apt-get install r2cloud --fix-missing
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
r2cloud is already the newest version (20240929180224).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 r2cloud : Depends: r2cloud-jdk (>= 17.0.5-1) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

Nammurg, 2024-10-03:
Hello, I have tried again today and it was successful... :?
I am now giving it another go with couple of passes, will let you know how it goes.

dernasherbrezon, 2024-10-03:
Yes, I fixed it couple of days ago. But currently working on the root cause fix.

Nammurg, 2024-10-05:
Thank you!
It seems that it is recording data, but I get no images...
Spectrogram looks like this for METEOR and the pass was 80 degrees.
[image]

dernasherbrezon, 2024-10-05:
r2cloud supports only LRPT protocol. The instrument that was sending the data via this channel failed and was disabled on METEOR-M 2 recently.

I disabled this satellite in r2cloud at 34c12c3a51aaa399ba4f8e4b7c51913e57f68a2b

dernasherbrezon, 2024-10-05:
Closing this ticket since the original issue seems to be resolved. Feel free to move into [discussions](https://github.com/dernasherbrezon/r2cloud/discussions) or telegram chat for further troubleshooting.

closed 2024-10-05 by dernasherbrezon