dernasherbrezon/r2cloud #148: R2cloud - Nginx.service failed

vk3ymml, 2024-12-02:
Hi

I have three raspberry pi on my network all seem to operate correctly.

I've installed r2cloud via the downloaded image. All seems fine. I configure wifi and reboot with no other changes and find that the nginx will not load therefore not able to configure or progress further.

I did at one point purge and reinstall nginx and rebooted it loaded however don't know what or where to go from that point.

Any suggestions would be appreciated

dernasherbrezon, 2024-12-02:
Hi,

It looks like r2cloud config got wiped with the nginx purge. Can you:

1. Purge and re-install r2cloud
2. Check nginx logs: journalctl -u nginx -n 200

dernasherbrezon, 2024-12-02:
Ok. So this is most probably fixed the issue. Feel free to join telegram chat @leosatdata or start new discussion in GitHub: https://github.com/dernasherbrezon/r2cloud/discussions

closed 2024-12-02 by dernasherbrezon